andreknieriem/photobooth #1: Tapping on QR icon is ignored on Android mobile device in gallery.

moritzklein, 2017-02-20:
If I tap on  the QR icon on Android mobile device in gallery, nothing happens. No QR is shown. On PC with chrome QR works just fine!

ben9978, 2017-03-04:
I have the same problem with iOS. Chrome/Safari/Firefox in macOS work fine.

moritzklein, 2017-03-06:
You need to edit the ".gal-qr-code" with a touch handler event:

$(document).on('click **touchstart**', '.gal-qr-code', function(e){

ben9978, 2017-03-07:
Hi Moritz, it is not clear to me where to make this modification. I suspected the .gal-qr-code in style.css. While this changed the appearance of the button, it didn't solve the problem. You don't have the solution in your forked version, do you?

moritzklein, 2017-03-07:
Hi Ben, no I have the solution on my raspberry :-). I have to push my source to my fork... havn't done it yet. Will try to do it this evening.

andreknieriem, 2017-03-07:
Fixed now.

closed 2017-03-07 by andreknieriem