andreknieriem/photobooth #2: Images are not centered if not having size 1920x1280

moritzklein, 2017-02-20:
To get correct image size for gallery, suggest to get correct image size for each image:

line 75 in index.php:
                    <?php
                    foreach($images as $image) {
                        $img_size = getimagesize("images/".$image);
                        echo '[image]
                                <figure>Caption</figure>
                            </a>';
                    }
                    ?>

andi34, 2019-06-04:
@andreknieriem https://github.com/andi34/photobooth/commit/d3fa9c55479f5c7ecf8d6650137b90a1f8448475

I've splitted F4bsi s commits to be more logical for a good history

andi34, 2019-09-05:
Should be fixed with https://github.com/andreknieriem/photobooth/commit/2cabe9f19045663ac7c0735a516fff20f739fbf3

closed 2019-09-05 by andi34