andreknieriem/photobooth #7: Feature suggestion: Guest download page instead of qr codes

vrs01, 2017-07-23:
Hi!

Instead of using the qr codes for download loading images I would like to download the images from gallery view. Since everybody needs to be connected to the photo booth Wi-Fi anyway, I think it is more convenient than the qr codes. One could implement also a password into the photo booth page so that without password, people can download images and the password is only entered on the tablet that is used for taking photos. One could even think of loading the photo booth page like a landing page right after connecting to the WiFi.

Best, vrs01

andi34, 2019-06-04:
+1 on downloading images directly from gallery.
Isn't it prepared already by the download.php?

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

andi34, 2019-10-11:
Reopen and close collection thread

commit c4ace13c (2019-11-14): Add download picture button to gallery

fix #7
closed 2019-11-14 by sualko