andreknieriem/photobooth #10: Feature request: Trigger photo with external button

vrs01, 2017-07-24:
I noticed that some people struggle to press the buttons on a touchscreen, because it needs to be the right length and pressure of the button push.
I would like to attach a physical button that triggers the photo shooting. Either by foot: https://www.amazon.de/KKmoon-Fußschalter-Keyboard-Aktion-Schalter/dp/B00DG0KQBE/ref=sr_1_1?ie=UTF8&qid=1500925768&sr=8-1&keywords=usb+taster
or by some 'hand button'
How can this be implemented?

Tomtaste, 2017-08-12:
My favourite is this one:
https://www.amazon.de/FU%C3%9F-GROBHANDTASTER-1S1%C3%96-FAK-R-KC11/dp/B0014DQI5S/ref=sr_1_1?ie=UTF8&qid=1502548311&sr=8-1&keywords=grobhandtaster
Any ideas how to implement?

wirthual, 2017-08-13:
Hi,

I think this is really helpful: https://github.com/PiPHP/GPIO

andi34, 2019-09-05:
Via GPIO possible now.
Via USB should be possible too if accesskey is definedhttps://github.com/andreknieriem/photobooth/blob/dc99f6dcf8465d994d8d5152bcf38edba8efa746/index.php#L68

closed 2019-09-05 by andi34