andreknieriem/photobooth #9: Feature request: Text overlay for photo printing

vrs01, 2017-07-24:
It would be nice to overlay one line of text maybe at the bottom of the printed fotos. Like the title of the event and the date ("Wedding of XY and YX, 1st of April 1916").

Seppel0815, 2018-06-26:
Look at imagemagick, you can edit the php Files (i have insert a overlay Watermark with imagemagick)

jensfr1, 2018-07-19:
> Can you share the example?

thymon13, 2019-02-26:
Yes I need too :)

vdubuk, 2019-09-06:
file   print.php   uses GD to merge the Photo and QR code

GD has a function to add text to an image   imagefttext     

I am a newbie with php but am going to  try to use GD imagefttext  as it is already used in print.php

vdubuk, 2019-09-08:
Very rough test.  in file print.php  after 

`imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));`

add the following 
`// text on print - ensure you include path to your font
imagettftext ($print, 30, 0, 850, 350, $black, 'resources/fonts/GreatVibes-Regular.ttf' , 'Line 1 text' );
imagettftext ($print, 30, 0, 850, 450, $black, 'resources/fonts/GreatVibes-Regular.ttf' , 'Line 2 text' );`

vdubuk, 2019-09-09:
My next step is to try and add options to the admin panel to allow for easier change of text, font, size etc..

andi34, 2019-09-09:
Thanks. Let me know if you have something ready, else I might be able to look at it at some point this week.

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

closed 2019-09-09 by andi34