andreknieriem/photobooth #12: gphoto2 : How to add an option ?

thymon13, 2017-08-21:
Hi,

First, thanks a lot for this amazing Photbooth, it's so great !!
But I've a question :)

Where can I add this option : 

> --keep-raw
> When doing --capture-image-and-download or interval capture, this option will keep the RAW images on the memory card of the camera, but still download the JPEG images. This is useful when doing dual mode capture and you want to review the JPEGs already during capture.

Thank you in advance.

Thymon

andreknieriem, 2017-08-21:
You can take a look in the config.inc.php $config['take_picture']['cmd'] = 'sudo gphoto2 --capture-image-and-download --filename=%s images'; Here you can add an option.

closed 2017-08-21 by andreknieriem

thymon13, 2017-08-21:
I Know...
But this doesn't work :(

`$config['take_picture']['cmd'] = 'sudo gphoto2 --capture-image-and-download --filename=%s images --keep-raw';`

thymon13, 2017-08-23:
You must Add this option  : --set-config capturetarget=1

[Here](https://github.com/gphoto/gphoto2/issues/90)

andreknieriem, 2017-08-23:
Ah thanks for the hint