andreknieriem/photobooth #13: Feature Request - Optionpage

HerrVonundZu, 2017-09-10:
Hey there
it would be really helpfull to have an admin/optionspage where you can deactivate/activate the qrcode/printbutton, reset fotos/data/thumbnails etc...

andreknieriem, 2017-10-02:
Yes, thats true. Will try to add it soon.

soko1988, 2018-07-31:
Hey @andreknieriem,
great Project!
Possible to Secure the Adminpage with a PIN-Code or RFID-Chip?
Also for Issue #9 : Configure/Upload the Overlays (Text / Image) and maybe set the place for that overlay.

Tiptronic1812, 2019-01-05:
Hey there, is the admin/optionpage already implemented? Where can I find it?

andreknieriem, 2019-05-15:
Hi, small update. Was working on it. Think it will be ready soon.

http://photobooth.andrerinas.de/admin/

ChrisKram, 2019-05-16:
Looks great!

andreknieriem, 2019-05-18:
Now availabled

closed 2019-05-18 by andreknieriem