andreknieriem/photobooth #14: Feature Request - Live/Preview

HerrVonundZu, 2017-09-10:
it would be nice to have a preview 
in the comments on your website a guy wrote something about inserting a motion stream as backgroundpicture. Maybe it is possible to show this webcam/raspberrypicamstream in a tiny frame on the startscreen and as huge backgroundvideo when the countdown starts.

TKnpl, 2017-11-29:
Would be a great enhancement! I would also be very interested in this extension.
If possible, a new part in the configuration file will be nice, to disable/enable the live-view while the countdown is running.
Also streaming the live-view from the dslr-camera via. gphoto2 or ffmpeg would be great. But I think this requires a serverside service... hopefully raspberrypi is powerfull enough.

andreknieriem, 2019-05-18:
Is now via device webcam availabled. Just activate in /admin

closed 2019-05-18 by andreknieriem

ChrisKram, 2019-05-19:
Is that only an Option if I use a tablet with a camera or will it also work with a Raspberry Pi camera?

UncleCCC, 2019-09-23:
Läuft das nur mit der Tablet Kamera?
Oder kann ich zB eine Webcam für LiveView verwenden?

UncleCCC, 2019-09-23:
Ah, sry, bin in sachen Programmierung nicht soo Fit..
Aber sehe ich es richtig das MotionEye für den Stream sorgt?

andi34, 2019-09-24:
@nie-oh might be able to give proper information about he setup on his pi with USB webcam