andreknieriem/photobooth #19: Update installation procedure:

DigiOrti, 2017-12-30:
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install git apache2 php php-gd gphoto2 libav-tools
cd /var/www/
sudo rm -r html/
sudo git clone https://github.com/andreknieriem/photobooth
sudo mv photobooth html
sudo chown -R pi: /var/www/
sudo chmod -R 777 /var/www

Seilafeh, 2018-01-07:
+1 der dev Modus in der config ist fies :-D Habe mir ewig den Kopf zerbrochen, wieso die Kamera nicht auslöst, bis ich die Option entdeckt habe.

thymon13, 2019-02-26:
> So please search for a USB connection and use the CP910 printer driver in CUPS.

How please ?

thymon13, 2019-05-11:
I use Canon CP1300 but I can not launch a second impression until the previous one is finished. Otherwise the printer crashes ... :(

Do you know how to do it ?

closed 2019-06-07 by andreknieriem