andreknieriem/photobooth #22: Create responsive design

mhellmeier, 2018-02-28:
The UI looks really nice on a desktop pc. I tried it on my phone and some tablets but it looks bad there. I want to use the webinterface with a tablet in the real fotobox.

**Design Problems**:
- [ ] "Creating foto"-button in the middle of the text on the front page
- [ ] Width / Height of the hole site is wrong (I can scroll left / right or up / down on my mobile devices)
- [ ] Gallery to small or rather only on a half screen
- [ ] QR Code to big on mobile devices in landscape mode
- [ ] Various bugs after rotating the phone / tablet

BrixSat, 2018-02-28:
Thanks :) will try to improve it. 

Can you please share the tablet model and a real picture of the photo-booth?

Thanks in advance.

andreknieriem, 2018-02-28:
Hey BrixSat, thanks for your help! Will you add a pull request?

mhellmeier, 2018-02-28:
> Can you please share the tablet model and a real picture of the photo-booth?
Yes of corse! I am using a Nexus 7 tablet. The photo-booth isn't finished so I can't show you a real-world picture of it. But I want to put the Nexus 7 tablet in landscape mode into the box. I will add some screenshots - hope it helps!

> The browser you are using on the tablet?
I am using Google Chrome (shipped with Android)

You can also reproduce the design failtures with the Google Chrome desktop version.

[image]

[image]

[image]

F4bsi, 2018-03-11:
@BrixSat 
Hey, 
I couldn't find any changes to the design in your repository, so I started to work on it. I hope that's ok? :-)

BrixSat, 2018-03-11:
Go for it :)

F4bsi, 2018-04-01:
Submitted #24 to fix this issue. 
Maybe you could try it out?

andi34, 2019-09-11:
You could cherry-pick:
https://github.com/andi34/photobooth/commit/d317f58d95752ac585c8e2545ef24957ceb11a4c

https://github.com/andi34/photobooth/commit/9dab3dfae43342952ba5f6db85c5123aa3869c0b

And

https://github.com/andi34/photobooth/commit/250172ae175e6f815375335dcce566b840ad7bb2

Hope it helps.

PR #78 (2019-09-27): responsive layout
This is work in progress, but the startscreen and the gallery is already responsive.

fix #22 

[image]
[image]
closed 2019-09-27 by andi34