andreknieriem/photobooth #23: Relative path

mhellmeier, 2018-02-28:
You can only install photobooth into the root directory of your apache webserver (`/var/www/`). But I want to install photobooth into a subdirectory (like `/var/www/photobooth/`). This doesn't work because the code trys to jumps directly into the hard-coded folders / files.

One solution would be creating a config variable with the path of the application (in my example this would be `/photobooth/`). Then put the config variable in front of every include or resource.

After trying to install photobooth into a subdirectory, I got the following result:
[image]

BrixSat, 2018-02-28:
Yes i am aware of that, going to be fixed soon :) i have a new release to
be made in the next 2 months or so (as soon as it works nicely).

Thanks for the help :)

andi34, 2019-09-25:
Submitted: https://github.com/andreknieriem/photobooth/commit/ec42bc52c15c5645044f40841b6665e41b333047

closed 2019-09-25 by andi34