andreknieriem/photobooth #29: Get images from picamera

buchey8070, 2018-07-22:
Hey,
I want to get my Pi camera module get working with your photo booth code. 

Already adapted the config.inc.php for cmd to look like:
`$config['take_picture']['cmd'] = 'sudo raspistill -n -t 10 -o %s';`

The photos get taken, but aren't displayed on the webpage "An error occureed, please try again".

vdubuk, 2019-09-02:
Hello to everyone.  First of all, a big  thank you to all those who have contributed towards this project.

@buchey8070 Did you manage to get the Pi Camera to work?

andi34, 2019-09-02:
You could try ```$config['take_picture']['cmd'] = 'sudo raspistill -n -t 10 -o /var/www/html/images/%s';```

vdubuk, 2019-09-02:
Hi  @andi34 

I had tried the original code and also your code but I just get the following on screen message when trying to use the photobooth.   

Something went wrong please try again

I tried the code in terminal and it will save a photo to the images folder if i substitute %s for test.jpg

> You could try `$config['take_picture']['cmd'] = 'sudo raspistill -n -t 10 -o /var/www/html/images/%s';`

andi34, 2019-09-02:
You've tried what @andreknieriem suggested?

andi34, 2019-09-02:
Could you please run ```sudo raspistill -n -t 10 -o /var/www/html/images/%s``` attach a screenshot here taken after running the command?

vdubuk, 2019-09-04:
> Could you please run `sudo raspistill -n -t 10 -o /var/www/html/images/test.jpg` attach a screenshot here taken after running the command?

Adding suggested code to Admin of webapp and running the app gives an error and no image is generated.
[image]

Running command directly in terminal generates a jpg 
[image]

vdubuk, 2019-09-04:
contents of  var/log/apache2

[Wed Sep 04 10:44:17.324158 2019] [php7:notice] [pid 525] [client 127.0.0.1:46128] PHP Notice:  Undefined variable: config in /var/www/html/folders.php on line 4, referer: http://127.0.0.1/
[Wed Sep 04 10:44:17.324508 2019] [php7:warn] [pid 525] [client 127.0.0.1:46128] PHP Warning:  Invalid argument supplied for foreach() in /var/www/html/folders.php on line 4, referer: http://127.0.0.1/
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

I have tried increasing the GPU mem to 256 and also done rpi-update

andi34, 2019-09-04:
```
$config['take_picture']['msg'] = '';
```
Should be enoty

vdubuk, 2019-09-06:
I understand where I was going wrong. 

I cannot use the PiCamera to 'see preview by device cam' and use PiCamera to take pic.

andi34, 2019-09-10:
Pi Camera funktioniert mit folgenden Codezeilen in der config.inc.php
```
$config['take_picture']['cmd'] = 'sudo raspistill -n -o $(date +%s) | echo "Done"';
$config['take_picture']['msg'] = 'Done';
```
raspistill erzeugt keine Ausgabe nach der Bildaufnahme daher muss man mit echo tricksen.

English:
Pi Cam works with mentioned config changes.
Raspistill does not give any feedback after the picture was taken, workaround for that with "echo".
Thanks to Andreas Maier for that information

chris883773, 2019-10-25:
i have still this problem and get the error: 
File was not created.
If i use this 
sudo raspistill -n -o $(date +%s) | echo "Done"
at shell i take a image and earn the Message done. 

in my.config.inc.php i got  the following one:

<?php
$config = array (
  'take_picture' =>
  array (
    'cmd' => 'sudo raspistill -o $(date +%s).jpg -q 100 -t 1 | echo "Done"',
    'msg' => 'Done',
  ),
);

i want to use the Raspicam v2 with a touch monitor and a canon printer.

andi34, 2019-10-25:
```
$config['take_picture']['cmd'] = 'raspistill -n -o $(date +%s) | echo Done';
$config['take_picture']['msg'] = 'Done';
```
https://github.com/andreknieriem/photobooth/wiki/FAQ#is-pi-camera-supported