andreknieriem/photobooth #32: Kamera löst nicht aus

sburger83, 2018-08-08:
Hallo,
Ich benutze einen RaspberryPI 3 und eine Canon DSLR.
Bei klick auf neues Bild wird die Kamera nicht ausgelöst. Steuere ich die Kamera direkt über gphoto an macht sie ein Bild.
wo soll ich nach dem Fehle suchen?

F4bsi, 2018-08-09:
Hey,
Kann es sein dass du den Developer Modus in den Einstellungen noch an hast?
Bei mir funktioniert es ohne Probleme mit einer Canon DSLR...

sburger83, 2018-08-09:
Wo kann ich den denn ausschalten? Leider funktioniert die Seite mit dem Tutorial nicht, ansonsten hätte ich da nachgeschaut.

F4bsi, 2018-08-09:
In der config.inc.php Datei im Hauptverzeichnis musst du die Zeile $config['dev'] = true zu $config['dev'] = false ändern. Denn sollte es funktionieren

soko1988, 2018-08-10:
@sburger83 Das Tutorial ist gut beschrieben. Da steht auch ganz deutlich, dass man den DEVELOPER Modus deaktivieren soll.

andreknieriem, 2018-08-10:
Hey die Seite läuft wieder. Es gab einen Fehler im Firefox Browser mit einem JS.
https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html

closed 2018-08-10 by andreknieriem