andreknieriem/photobooth #34: Foto wird zweimal ausgedruckt

Norman-Sch, 2019-01-05:
Wenn von der Home-Seite ein Druckauftrag abgeschickt wird, dann wird das Bild zweimal gedruckt. Druckt man aus der Gallerie so wird nur ein Druckauftrag angelegt.

dArkjON, 2019-06-04:
Hi Guys,

ive the same problem when i make a picture and press the print button via touchscreen it try to print 2 pictures. When i use a normal mouse and press the print button only one print is in the line.

I have try to capture the second click with a alertbox but it still send the print job 2x.

Norman-Sch, 2019-06-04:
I could solve this problem this way:
In the file photobooth/resources/js/core.js
in row 126 and 127
Replace 'click touchstart' with 'click touch'

I have a Medion Tablet and use Chrome Browser but It was the same with other browsers

Das hat wohl mit der Toucheingabe auf dem Tablet zu tun...

Nie-Oh, 2019-07-31:
ich habe in der cups config eingestellt, das immer nur ein Druckauftrag in der Warteschlange sein darf.

devnull1975, 2019-07-31:
In der cupsd.conf oder über die Adminpanel von Cups oben die Zeile "MaxJobsPerPrinter" einfügen (ohne die "").
Wenn du das über die Adminpanel machst startet der Druckserver neu. Über die Kommandozeile musst du "sudo Service cups restart" eingeben. 
Ich hab zusätzlich noch "MaxJobsPerUser" drin stehen. Welcher genau nun dafür sorgt das nur ein Bild gedruckt wird eh ein zweiter Auftrag angenommen wird weiss ich nicht. Stört aber auch nicht weiter.

andreknieriem, 2019-08-29:
Sollte jetzt behoben sein. Es waren 2 Events im Javascript. Touchstart und Click. Manche Browser meinen aber dass ein Touch auch gleichzeitig ein Klick ist und schicken dann die Funktion 2 mal ab.

closed 2019-08-29 by andreknieriem