andreknieriem/photobooth #35: Take 10 second videos

tobiasHaecker, 2019-01-27:
Is it possible to take 10 second videos instead of a picture?
Which file do I have to modify?
Thank you for your support.

ChrisKram, 2019-05-05:
What about the command:

`[--capture-movie SECONDS or COUNT]`

but I can't find an option to download them. Maybe in combination with:

`--wait-event-and-download SECONDS, MILLISECONDS, COUNT or MATCHSTRING`

gphoto documentation means that it requires a newer canon or nikon camera.

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

andi34, 2019-10-11:
Reopen and close collection thread

rawbertp, 2019-11-04:
I like that idea. I might be looking into it if I find some time. The easiest way of implementing it is to use `--capture-movie SECONDS` which should work for all live-preview-capable DSLRs. This is with no sound, obviously. If sound was an requirement, it gets more complicated as Nikon and Canon cameras behave different (if I get the documentation right, it doesn't really work for EOS cameras at all, as you would have to switch them to video mode manually).

dennis0307, 2020-03-20:
Hi guys,

I am far away from having the skills to do this on my own but I was also looking for possibilities to do GIFs on a raspberry based photobooth. I am using this photobooth for quite a while and I am more than happy but having the GIF possibility would make it perfect.

I found another webpage with a tutorial: https://www.instructables.com/id/Raspberry-Pi-Tumblr-GIF-Photo-Booth/ where python code was used to take a series of 6 pictures and convert it to a video file (GIF) afterwards. I am quite new to python but the code looks pretty straight forward for taking the picture so it might be possible to integrate it in here.

If someone with more coding skills can have a look into this, it would be awesome!

Greetings from Germany :)