andreknieriem/photobooth #36: the camera triggers, but nothing

thymon13, 2019-02-24:
Hello,

Firstly sorry for my english.
I had version 1.2.0 that worked perfectly.

I installed the latest version 1.3.2 on another Raspberry. And now it does not work.
The camera is shooting well, but "something went wrong" and the picture has become unreadable on the CANON 7D.

How can I find where the problem is? Is there a debug mode?

Thank you in advance for your help.

thymon13, 2019-02-25:
OK i found... I'm noob :) :)

closed 2019-02-25 by thymon13

andreknieriem, 2019-02-26:
Ok. Can you tell me what the problem was?

thymon13, 2019-02-27:
Hello,

I think that all the following commands did not work.
`sudo apt-get install git apache2 php5 php5-gd gphoto2 libav-tools`
And I did not pay attention ...
Probably because of Php5 that no longer exists as is.
It would be necessary to modify the order
`sudo apt-get install git apache2 php php-gd gphoto2 libav-tools`