andreknieriem/photobooth #38: Feature request: Hotfolder and only use the Gallery

alpha-io, 2019-05-19:
Hotfolder to import new Images. I want to Capture the Images with another program, to get real Live View from DSLR, but i love the Gallery from photobooth+.

So i only want to use the Gallery.

ChrisKram, 2019-05-19:
What does hot folder mean in your case? The gallery picks up the pictures where gphoto put them in. So you may just copy the pictures into the images Folder of photobooth or advice you other Program to save them into var\html\Images.

In the next step you disable all buttons on the start page except of the Gallery and use the page to only to view images. The page will stay in the Gallery Mode as long as someone close it.

alpha-io, 2019-05-19:
I tried that already, but it's not working. 
Only when i add the filenames manualy to the data.txt file.

MrMaavin, 2019-05-20:
You need to add the name of the pictures in the the `array` which in in `data.txt`. The gallery get's the image names from this file. But you also need to add them in the `var\html\Images` folder.

andreknieriem, 2019-05-20:
That's not a feature I will add. But you can look here:
https://github.com/andreknieriem/photobooth/blob/master/admin/admin.php#L12

This is a PHP command to list all jpg files in a given folder. You can use this as $images variable and then the gallery should work.

closed 2019-05-20 by andreknieriem