andreknieriem/photobooth #39: Feature Request - Protection of the Admin Page

TKnpl, 2019-05-19:
Hallo Andre,
vielen Dank, dass du dieses coole Projekt der Öffentlichkeit zur Verfügung stellst!
Zu deinem neuesten Release 1.5.0 habe ich eine Anmerkung bzw. einen Verbesserungsvorschlag.
Derzeit ist die Admin Seite von jedem aufrufbar und die Inhalte manipulierbar, der diese Seite kennt.
Da bei der Download Option via. QR Code bereits die Base URL ersichtlich ist, kann ich mir vorstellen, dass die Admin Seite recht schnell gefunden werden kann.
Daher würde ich mir wünschen, dass die Admin Seite in der Konfig deaktivierbar ist und/oder bei Aufruf der Seite eine Passworteingabe erfolgen muss, sodass nicht jeder die Inhalte ändern kann.
Vielen Dank und viele Grüße

Seppel0815, 2019-05-19:
Eine Idee: 

Erstell doch eine .htaccess in dem Unterordner /Admin 

Lg

andreknieriem, 2019-05-20:
Ja das würde ich auch empfehlen, wenn du denkst, dass jemand die Admin Page suchen könnte. Das Photobooth ist ja eigentlich nicht via Internet erreichbar und dann müsste es schon jemand der Gäste sein. 
Evtl. erstelle ich noch eine .htaccess als Demo Datei in dem Ordner, die man benutzen kann, oder nicht.

https://tools.andrerinas.de/htpasswd_generator

andi34, 2019-10-11:
Reopen: https://github.com/andreknieriem/photobooth/pull/118

commit 6a88ce2e (2020-01-11): Merge pull request #118

admin: add simple login

Fix: #39
closed 2020-01-11 by andi34