andreknieriem/photobooth #40: Kamera löst nicht aus

abacaxi3000, 2019-05-20:
Hi zusammen,

obwohl das Tutorial ja wirklich einfach ist (denn ich bin ein absoluter Anfänger), schaffe ich es nicht, die Kamera zum auslösen zu bekommen. Auf der PhotoBooth Website lässt sich nur ein Screenshot erstellen...
Gphoto2 ist korrekt installiert, denn mit dem manuellen Befehl knipst die Kamera. Den Developer Mode habe ich ausgeschaltet. Und auch die Rechte für den Webserver-User "www-data" in etc/sudoers meines Erachtens gegeben. Ist es denn von Belang, an welcher Position sich die Zeile "www-data ALL=(ALL) NOPASSWD: ALL" in der Datei befindet?

Der Witz ist, ich habe auf einem geliehen Raspberry das ganze schon hinbekommen. Mit viel Try & Error, weil es meine ersten Schritte meinem Respberry waren und ich meine ich hätte mich auch ewig mit den Rechten gekämpft. Aber fragt mich nicht, was ich damals gemacht habe.

Ich hoffe auf etwas Unterstützung von euch...

Beste Grüße
Marcus

andreknieriem, 2019-05-20:
Hallo Marcus,
hast du php-gd installiert? Der Fehler kann kommen, wenn er kein Thumbnail für die Galerie erstellen kann. apt-get install php-gd oder php-7.2-gd oder je nachdem welche php version du hast.

abacaxi3000, 2019-05-20:
Hallo Andre,

danke für die rasche Hilfe. Ich habe nochmal php-gd installiert, aber daran lag es nicht. Das Programm hat auch vorher schon Thumbnails erstellt und auch regulär im Ordner thumbs gespeichert. Halt nur als Screenshot. 

Hättest du sonst noch eine Idee? Für mich wirkt es (als Laie) so, als würde doe Kamera nicht angesteuert werden...

abacaxi3000, 2019-05-20:
So sieht meine sudoers Datei aus. Ist da evtl etwas falsch?:
____________

#
# This file MUST be edited with the 'visudo' command as root.
#
# Please consider adding local content in /etc/sudoers.d/ instead of
# directly modifying this file.
#
# See the man page for details on how to write a sudoers file.
#
Defaults        env_reset
Defaults        mail_badpass
Defaults        secure_path="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

# Host alias specification

# User alias specification

# Cmnd alias specification

# User privilege specification
root    ALL=(ALL:ALL) ALL
www-data ALL=(ALL) NOPASSWD: ALL

# Allow members of group sudo to execute any command
%sudo   ALL=(ALL:ALL) ALL

# See sudoers(5) for more information on "#include" directives:

#includedir /etc/sudoers.d

Defaults        rootpw

abacaxi3000, 2019-05-20:
Weil ich es einfach nicht hinbekomme, habe ich das Raspbian nochmal neu aufgespielt und mich wirklich sklavisch Schritt für Schritt an die Anleitung gehalten und parallel dokumentiert. Siehe pdf Datei im Anhang.
Leider will es immer noch nicht laufen. Über die Eingabe der IP komme ich auf die PhotoBooth Seite, der Countdown läuft und dann wird wieder nur ein Screenshot gemacht.
GPhoto läuft, da der Befehl $ gphoto2 --capture-image  ein Bild aufnimmt. 

Ist es doch ein Problem mit den Rechten von www-data User? Hier auch nochmal die Frage, ob die Zeile "www-data ALL=(ALL) NOPASSWD: ALL" an einer bestimmten Stelle sein muss?

Beste Grüße

Marcus

[Vorgang_Protokoll.pdf](https://github.com/andreknieriem/photobooth/files/3199313/Vorgang_Protokoll.pdf)

andreknieriem, 2019-05-21:
Du installierst php5 extras, mitlerweile wird aber php7 automatisch installiert. Nimm mal die 5 in deinen befehlen raus. Also apt get install php-gd. Ansonsten lege mal ne test.php mit phpinfo() in den ordner und ruf die auf und guck welche php version installiert ist.

abacaxi3000, 2019-05-21:
Ok, vielen vielen Dank für den Support. Ich probiere das heute Abend aus und werde berichten.

abacaxi3000, 2019-05-21:
Es will einfach nicht. Ich hab nochmal alles neu aufgespielt um eine saubere Version zu haben. Deinen Rat, php5 nicht zu installieren hab ich befolgt und die Ziffer "5" aus den Befehlen genommen.

Mit dem Kommando $ php -v bekomme ich folgene Meldung:

pi@raspberrypi:~ $ php -v
PHP 7.0.33-0+deb9u3 (cli) (built: Mar  8 2019 10:01:24) ( NTS )
Copyright (c) 1997-2017 The PHP Group
Zend Engine v3.0.0, Copyright (c) 1998-2017 Zend Technologies
    with Zend OPcache v7.0.33-0+deb9u3, Copyright (c) 1999-2017, by Zend Technologies

Also sollte php 7.0.33 installiert sein? 

Trotzdem bleibt der Fehler bestehen... Die PhotoBooth App erstellt nur einen Screenshot. Mit dem direkten gPhoto Befehl löst die Kamera aus und wenn ich in der App auf "drucken" klicke, knipst die Kamera ebenfalls - ohne etwas zu speichern.

Hast du noch einen Rat?

Norman-Sch, 2019-05-21:
aber in der config.inc.php hast du $config['dev'] = false; gesetzt?
Bzw. alternativ auf der Admin Page den Entwicklermodus deaktiviert?

abacaxi3000, 2019-05-21:
Ja, das habe ich in der config.inc.php geändert und steht auf „false“. Hab ich auch mehrfach geprüft.

schwarzp93, 2019-05-21:
Hallo :)

Ich kann leider nicht helfen. Ich habe aber exakt das selbe Problem. Habe auch vor 2 oder 3 Tagen versucht die Fotobox zu installieren. Wurde vielleicht kürzlich irgendetwas verändert?

Viele Grüße,
Philipp

Norman-Sch, 2019-05-21:
Habt ihr es auch über die Admin-Seite geändert?

Die Datei admin/config.json ist wahrscheinlich relevant. Dort muss der Entwicklermodus ebenfalls auf false gesetzt werden.

abacaxi3000, 2019-05-22:
Hallo Norman,
nein habe ich nicht, weil ich mich an das Turorial gehalten habe und bei dem Versuch von mir, wo es lief, auch ziemlich sicher nicht in der admin/config.json etwas geändert habe.
Ich probiere es aber gerne aus, sehe nur nicht, wo ich dort den Entwicklermodus auf false stellen kann.

Ich bin zudem beruhigt, dass ich scheinbar nicht der Einzige bin, bei dem jetzt kürzlich dieses Problem auftritt...
Beste Grüße
Marcus

andreknieriem, 2019-05-22:
Hallo Marcus,
wenn du im DEV Modus wärst, dann würde er das Standard Bild einfach anzeigen, wie auf der photobooth Demoseite. Wenn der Fehler kommt, scheint es ein anderes Problem zu geben. Welche Kamera hast du? Ich kann mich erinnern, dass manche Kameras kein Bild machen wollen, bzw. nur eines machen und dann busy bleiben.

https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html

obkram, 2019-05-22:
Hallo zusammen,
Ich hoffe es fühlt sich jetzt niemand persönlich angesprochen oder provoziert denn das ist nicht der Hintergrund meiner Frage, aber...
@ Andre: hast du mal überlegt vielleicht für den RPI ein fertiges Image basierend auf dem RPI3b+ zum Download zur Verfügung zu stellen? Damit würdest du zumindest 80% der FAQ umgehen.

andreknieriem, 2019-05-22:
@obkram Ja darüber habe ich schonmal nachgedacht. Ich weiß aber nicht wie es mit den Rechten ist, da ja schon recht viel von anderen installiert wird. Ich kann aber nochmal gucken.

obkram, 2019-05-22:
@ Andre:
Ein frisch aufgesetztes Raspbian mit den Grundfunktionen deines Photobooth würde ja schon reichen. Darauf kann dann jeder aufbauen wie er es mag.
P.S.: Ich finde es toll das du deine Entwicklung noch so lange nach deiner eigenen Hochzeit weiterführst und die Leute immer noch supportest. Ein dickes DANKESCHÖN von meiner Seite. Deine Software hat uns allen auf so mancher Party schon sehr viel Freude bereitet.

abacaxi3000, 2019-05-22:
Ich habe eine Sony Alpha 6000. Beim Raspberry eines Kollegen hat alles funktioniert. War ebenfalls eine saubere Installation und ein PI 3 B+...
Von daher kann es an der Kamera nicht liegen?
Ein Image fänd ich auch interessant. 
Und dem Dankeschön, dass du dich sehr um eine Lösung bemühst, schließe ich mich auch an! Sowas ist nicht selbstverständlich.

ChrisKram, 2019-05-22:
Ich würde insgesamt mal versuchen, mit der Vorgängerversion zu arbeiten. Die JSON Datei hat es vorher nicht gegeben, deshalb musste man da auch nichts ändern. 

Bzgl dies Images, ich hab ein halbfertiges auf Basis von Raspian ohne Desktop, mit Desktop logon und Nginx. Bin aber mit Nginx noch nicht weitergekommen, wollte was Performanteres als den Apache. 

Auf dem Raspian lite hat man dann auch keine Zusatztools und kann sich bzgl der Lizenz auf wenige Applikationen beschränken. Das war doch mit Rechten gemeint, oder?

abacaxi3000, 2019-05-22:
Wie komme ich denn an die älteren Versionen?

Sorry, wenn ich immer wieder Rookie-Fragen stelle, aber ich bin noch blutiger Anfänger und arbeite mich erst nach und nach in die große Welt des Raspberry rein.

MrMaavin, 2019-05-22:
Einfach unter Releases hier: https://github.com/andreknieriem/photobooth/releases

abacaxi3000, 2019-05-22:
Wir haben die Lösung!!! 
ChrisKram hatte den richtigen Hinweis. Mit der Version 1.3.0 läuft alles so wie es soll! Vielen Dank.
Also muss irgendwas an der aktuellen Version nicht stimmen! 

Vielleicht kannst du den Hinweis auch ins Tutorial geben!

ChrisKram, 2019-05-22:
Als Lösung würde ich es nicht bezeichnen, eher als Workaround. 

Leider kann ich Version 1.5 aus Mangel an Raspis grad nicht testen. Vllt nächste Woche dann, wenn ich die Box zurück habe.

abacaxi3000, 2019-05-22:
Der Fehler wurde im Tutorial auch gefunden. Im Ordner /var/www/html/admin muss in der config "dev" ebenfalls noch auf "false" gestellt werden.

_______
Auch wenn es Off-Topic ist: Wie empfehlt ihr mir denn die Erstellung des lokalen WLANs (dumme Frage: Ist das ein Access Point?)?
Via Router oder dem eingebauten WLAN? Ich finde da 10000 Anleitungen, aber die zwei, die ich probiert habe, laufen nicht...

obkram, 2019-05-22:
Ich habe den RPI nach dieser Anleitung direkt als WLan Hotspot eingesetzt...
https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

obkram, 2019-05-22:
Den letzten Punkt ...
Using the Raspberry Pi as an access point to share an internet connection (bridge)
... habe ich allerdings weggelassen. Ich lasse die Box autark laufen.

ChrisKram, 2019-05-22:
Ich hab für das WLan sowie von André beschrieben einen Router verwendet. Bisher ein ungenutztes Airport Express und letztes WE habe ich mir einen Mini WLan Router von TPLink gekauft so wie einen LTE Router der eine mobile Internetverbindung aufbaut und WLan aufspannt, da ich plane die Fotos in die Cloud zu laden und dann diesen Link zu teilen, da mir nicht gefällt, dass der QR Code die Root Adresse preis gibt und man so auch vermeiden kann, dass die User in das Wlan der Box gehen müssen. Bisher existiert aber nur die Idee, einen Ansatz hab ich nur ganz grob. 

Ein Router is die komfortabelste Lösung fürs Wlan da dee Einrichtungsaufwand marginal is. Einfach das Gerät wie zu Hause nutzen, es ist kein Accesspoint, der braucht immer ein existierendes Netz. 

Zudem nutze ich das Wlan für einen PiNano der per einfachem Framebuffer und Samba eine Slideshow mit den erstellten Bildern macht.

andreknieriem, 2019-05-23:
Ok, ich schließe das Ticket dann mal :) Gut dass es bei euch klappt. Ich werde die Anleitung sowie die Version 1.5 noch etwas überarbeiten.  Danke fürs Feedback

closed 2019-05-23 by andreknieriem