andreknieriem/photobooth #41: Bild-Ordner können geändert werden - Bilder werden aber nicht angezeigt

Norman-Sch, 2019-05-21:
Wenn ich auf der Adminseite z.B. den Bild-Ordner nach /home/photobooth/images ändere dann werden die Bilder dort zwar abgespeichert aber weder auf der Home-Seite, noch in der Gallery angezeigt.

andreknieriem, 2019-05-22:
Ok werde ich mir ansehen.

andreknieriem, 2019-05-23:
Ich habe es erst einmal behoben. Allerdings geht das Ganze nur für Pfade ohne / vorne. Also images, aber nicht /photobooth/images, da das Javascript sonst nicht weiß, wo er gucken soll. Ich werde mal sehen, wie man das verbessert

closed 2019-05-23 by andreknieriem

Norman-Sch, 2019-05-23:
Das wäre super. Ich nutze ein Netbook mit Xubuntu 16.04 und möchte ungern die Systempartition mit Bildern füllen. Vielen Dank für deine Bemühungen!

andi34, 2019-06-13:
Symlink?

andi34, 2019-06-13:
cd /var/www/html
rm -rf pictures
ln -s /home/photobooth/images ./pictures

andi34, 2019-06-26:
Geklappt?