andreknieriem/photobooth #43: Live View einstellen + kleine Details im CSS

abacaxi3000, 2019-05-29:
Hallo Andre & Community.

Ich habe ja den letzten Fehler beheben können, die Box ist gebaut und eigentlich alles bereit für die Hochzeit.
Für den Feinschliff habe ich noch zwei, drei Sachen, bei denen ihr mir hoffentlich auch weiterhelfen könnt:
1) Fotos aus dem Ordner löschen. Da sieht man wieder, dass ich noch ein Raspbian Greenhorn bin. Was muss ich einstellen, damit ich die Testfotos aus dem Image und Thumps Ordner löschen kann?

2) Cheese: Kann ich den Schriftzug irgendwo ändern? Wenn nicht, wäre das nicht so schlimm....

3) Kann ich den Drucken Button komplett entfernen? Ist ohnehin nicht vorgesehen...

4) Live View:
Habe die Lösung von den Pull-Requests mit Motion genommen (https://github.com/andreknieriem/photobooth/pull/20). Jedoch bin ich aus zweierlei Gründen noch nicht zufrieden: Ich fände es gut, wenn die Vorschau permanent wäre. Bei mir startet sie erst mit dem Countdown und dann braucht die Webcam (nutze die Logitech C270) auch zwei Sekunden bis sie aktiv ist.
Könnte man das so einstellen, dass Sie einfach die ganze Zeit läuft?
Oder alternativ den Countdown auf 7 - 8 Sekunden verlängern?

Zudem scheint die style.css von Seilafeh, der das Live-View-Ding programmiert hat, nicht korrekt zu sein. Ich kann nämlich nicht auf die Gallerie zugreifen. Wo müsste das denn eigenstell werden? 
Ich habe die Zeilen aus dem folgenden style.css kopiert:

https://github.com/andreknieriem/photobooth/pull/20/commits/1e14f4a13f69bcd7a033d40111e958a063dba9bb

Wäre große Klasse wenn ihr mir wieder helfen könntet! 

Beste Grüße aus Münster

Marcus

Norman-Sch, 2019-05-29:
3. Den Drucken Button kannst du in der config.inc.php deaktivieren:
$config['use_print'] = false;

andreknieriem, 2019-06-07:
1. Geh auf deine Photobooth-Seite und gehe in /admin/. Dort kannst du alles einstellen und oben rechts auch zurücksetzen. ACHTUNG danach gehen Ändeurngen an der config.inc.php nicht mehr, da der Admin-Bereich die Datei überschreibt.

closed 2019-06-07 by andreknieriem