andreknieriem/photobooth #44: print confirmation

thymon13, 2019-06-01:
Hi,

Is it possible to confirm with pop up or other when I click on print button ?
Currently nothing happens on the screen when you click on it. And people by reflex reclick several times over.

Sorry for my English :)

Thanks in advance.

thymon

thymon13, 2019-06-24:
Some help ?

tobiasHaecker, 2019-08-14:
Hallo André,
ich habe das so in die aktuelle Version eingebaut, funktioniert aber leider nicht.
Kannst du mir noch einen Tipp geben, was ich vergessen habe?
Vielen Dank.

andi34, 2019-09-02:
Doofe Frage: ```$config['print']['msg'] = '';```
Ist das schon implementiert nur eben keine Rückmeldung, da kein Text?
@tobiasHaecker kannst du Mal schauen ob es reicht
```$config['print']['msg'] = 'Druckauftrag gesendet....';

ChrisKram, 2019-09-03:
Im Admin Panel lässt sich ein Text eintragen, jedoch wird er nicht angezeigt. 

 @andi34, beziehst du dich mit dem Versuch auf die my.config.inc.php?
Bei mir wird die Datei nicht geändert. Ich habe nur die json Datei im Admin Folder, welche die Einstellungen enthält.

andi34, 2019-09-04:
Kannst du auch im Admin Panel eingeben in dem Fall.

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

closed 2019-09-09 by andi34