andreknieriem/photobooth #46: QR-Code from Gallery does not work

Norman-Sch, 2019-06-04:
QR-CODE from home does work. From gallery not. There is something wrong with the filename. I could not find the reason but i thing there is something wrong in the core.js.

File Name from home (works): http://192.168.178.47/download.php?image=20190603_212734.jpg

File Name from Gallery (works Not): http://192.168.178.47/download.php?image=images/20190603_213006.jpg

ChrisKram, 2019-06-05:
Da wird auch ein anderer Pfad für das Bild angeben, daher kann das nicht funktionieren. 

Oben ist es image=<filename>
Unten ist es image=images/<filename>

Da passt wohl eine Variable nicht.

Norman-Sch, 2019-06-05:
Ja es sieht so aus, als wäre es durch ändern einer variable einfach zu lösen, nur steige ich in der core.js  nicht durch. Vielleicht liegt das Problem aber auch in Photswipe oder phpqrcode.

andi34, 2019-06-05:
Kannst du mal schauen, ob das es behebt? https://github.com/andi34/photobooth/commit/21554da04aedd1e04678626265baae6dc1d90b17

Norman-Sch, 2019-06-05:
Habe beide Varianten probiert. Keine funktioniert, weder von der Home Seite, noch aus der Gallery.
Habe es wieder auf Original geändert. Damit klapppt der Download zumindest von der Home Seite.
Trotzdem danke für deine Bemühungen!

andi34, 2019-06-06:
Mh... Ich schaue die Tage gerne nochmal ob ich was finde. War die einzige Stelle die ich auf Anhieb gefunden habe wo der Pfad nicht per config läuft.

ChrisKram, 2019-06-06:
Ich denke, dass die Ursache in Zeile 288 der core.js Datei liegt. Korrigiert mich wenn ich falsch liege, aber ich denke da wird die Variable für filename für die qrcode.php erweitert um den Folder.

```
var img = pswp.currItem.src;            
img = img.replace('/'+imgFolder+'/', '');
$('<img>').attr('src', 'qrcode.php?filename=' + img).appendTo(pswpQR);
```

Ich hab jetzt noch nicht untersucht woher der QR Code den Unterordner kennt, aber kommentiere die Zeile 288 in der core.js einfach mal aus. Die Zeile ersetzt den Inhalt von img (pswp.currItem.src) mit images/pswp.currItem.src.

`//img = img.replace('/'+imgFolder+'/', '');`

andreknieriem, 2019-06-07:
Hab den Fehler behoben. Neue Version ist unterwegs ;)

andreknieriem, 2019-06-07:
Fixed in 1.5.2

closed 2019-06-07 by andreknieriem