andreknieriem/photobooth #47: libav-tools unavailable for ubuntu 18.04

developehrs, 2019-06-07:
Going thought the instructions. All well until post-processing the picture and showing it on the 
main screen. Also, there are no pictures in The Gallery but the pictures are saved in "images" locally on the computer, as expect. 

emil@wedding:~$ uname -a
Linux wedding 4.18.0-21-generic #22~18.04.1-Ubuntu SMP Thu May 16 15:07:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
emil@wedding:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

andi34, 2019-06-08:
https://launchpad.net/ubuntu/bionic/amd64/libav-tools/7:3.3.4-2

andi34, 2019-06-14:
@developehrs solved?If yes: how?

developehrs, 2019-06-14:
@andi34  Thank you for reminding me to update the post.

I would state that the issue got solved by restarting the computer and camera. The issue did, however, show up again but this was mainly due to human error and me trying to trick the system a bit too much. 

Summary:
Problem Solved
Issue was solved by restarting computer and camera.

closed 2019-06-14 by developehrs