andreknieriem/photobooth #49: White page after saving settings via admin panel

andi34, 2019-06-10:
Once I change settings via admin panel Photobooth main page doesn't work anymore. Deleting the config.json in admin folder makes it work again, but no config changes made after that.

andi34, 2019-06-10:
guess it's because config.json vs config.inc.php ...
Is the admin panel atm usable?

andreknieriem, 2019-06-11:
Pretty strange bug. I will take a look.

andi34, 2019-06-11:
I also tried deleting the config.inc.php instead, but also doesn't make it work

andi34, 2019-06-11:
While at it: an idea why admin Page isn't available after https://github.com/andi34/photobooth/commit/d3af99544d013834f9eb55d0734ab0cf6ad8d54f ?

Photobooth picks right config.

andreknieriem, 2019-06-12:
The config.inc.php is needed. Yo can't delete it! Maybe you have a syntax error in the config.php file? Is there a ";" missing or something like that? Maybe put the input in a php validator if your editor does not show php syntax errors.

andi34, 2019-06-12:
I'll check this evening. Thanks :)

andi34, 2019-06-12:
Mh , can't find anything wrong... Once I click save on admin Page I can't access the booth page to take a pic.  Admin Page still works.
Used my private branch and the dev branch which has the Pullrequest here. Same on both.

andi34, 2019-06-13:
I'll recheck in the evening if it's because of the changes of my PR as $config['file_format'] and depending configs are not controlled by admin Page. I'll add them to admin page in the evening, did some preparation already.

closed 2019-06-13 by andi34

andi34, 2019-06-13:
Problem had to do with the changes to add more configs to admin panel. All issues i had previously seem to be solved :)