andreknieriem/photobooth #55: Zweite Webseite für nur Gallery

koenigd79, 2019-07-04:
Hallo,

ich lasse die Photobooth auf einem am Raspberry PI angeschlossenen Touch Display 13,3" laufen. Im Kasten ist auch noch eine alte Fritzbox, so dass die leute via QR Code ihre bilder runterladen können.

Jetzt war meine Überlegung, ob man in der index.php nicht nur die Gallery hat, so können die Leute auch auf den Handy durch die Gallery surfen. Wenn ich die jetzt aber auf die normale index.php lenke, dann können sie ja auch die Kamera auslösen, was ich doof fände.

Daher Idee:
index.php ->zeigt nur Gallery (hier könnte auch dann ein direkter download button für die Bilder ergänzt werden)

photo.php -> zeigt die alte index.php an

Die neue Admin Seite ist super, ggf. könnte man noch in einer Ecke einen "verstecken" Link darauf machen, dann errreicht man die auch wenn man den Raspberry im Kioskmode betreibt.

Leider sind meine Programmierkentnisse nur rudimentär um ein solches Vorhaben umzusetzten, habe da schon versuche unternommen, aber alles zuerschossen...

Vielleicht gefällt die Idee ja noch jemandem und kann Sie umsetzten.

Zum Schluss möchte ich noch erwähen, dass dieses Projekt super ist und wir schon viel Spaß mit der Photobox hatten.

Gruß Daniel

andi34, 2019-07-07:
Die Idee mit der Galerie Seite finde ich auch gut. Schaue ich mir gern wenn ich Zeit finde Mal an, wenns bis dahin noch niemand umgesetzt hat.

Admin Bereich sollte nicht direkt von der Fotobox Seite erreichbar sein per Button, jedenfalls nicht ohne Passwortabfrage. Soll doch keiner an den Einstellungen Spielen bzw sollte auch keiner somit die Box zurücksetzen und vermutlich noch alle Bilder lösche dabei ;)

MoJou90, 2019-07-16:
Ich habe die Box auch auf einer Hochzeit aufgestellt. Kam super an. 

Als kleine Anpassung (Da ich keinen Touch-Screen habe) habe ich einen Taster angeschlossen, über den die Gäste die Kamera auslösen konnten. (Der Taster wurde über ein Python-Skript ausgelesen)

Allerdings war es so natürlich leider nicht möglich die QR-Codes anzeigen zu lassen. Ich habe dafür das WLAN freigegeben und jeder konnte auf die Galerie zugreifen. Leider ist die Seite auf dem Handy nur im Querformat zu nutzen. Es fehlte eigentlich ein Button um ein Bild aus der Galerie herunterzuladen. 

Meine PHP-Kenntnisse waren dann leider zu sehr eingerostet, als dass ich schnell einen Button hätte einbauen können.

Gruß

andi34, 2019-07-16:
@MoJou90 an dem Download button habe ich mich auch schon versucht... Ohne Erfolg bisher. Bin gerade dabei meine PHP Kenntnisse zu bekommen. Aber vllt hast du ja ne Idee was in meinem Code Snipped noch falsch läuft, habe ich hier gepostet https://github.com/andi34/photobooth/issues/4

ChrisKram, 2019-07-16:
@MoJou90 die Umsetzung mit dem Taster würde mich trotz Touchscreen ebenfalls interessieren. Das scheint mir intuitiver als der Druck auf den Screen.

Eine zweite Seite für die Galerie würde mich selbst ebenfalls interessieren, da ich sie gern auf einem anderen Bildschirm anzeigen lassen möchte, wo die Leute stöbern können, ohne das sie die Box blockieren und man ggf. auch dan en Drucker anschließen kann.

Bzgl. des Bildspeicherns. Ich hab das noch nicht ausprobiert, weil bei mir die Galerie gänzlich ausgeblendet ist, aber das Speichern eines Bildes sollte über das Kontextmenü des Browser funktionieren. Am Handy entsprechend lang gedrückt halten und dann Bild speichern wählen. So fern das die Seite nicht unterbindet, sollte es gehen. 

Im Prinzip muss der Button zum Speichern auch nichts anderes können, als das Bild außerdem halb der Galerie anzuzeigen, sprich "Bild anzeigen in neuem Tab". Dazu kann der Link verwendet werden, den der QR Code erzeugt oder besser gesagt, der dafür verwendet wird, den QR Code zu erzeugen. Wurde in #46   erläutert.

MoJou90, 2019-07-16:
Ich werde den Code hier Posten.
`code`
Dazu muss ich allerdings erst einmal mein dazugehöriges Image wiederherstellen. 
Habe es nach den Ausführungen von "Martin"
https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html
umgesetzt. 
Ist recht simpel gehalten. 
In der index.php wird ein hotkey für die Betätigung des Buttons zur Aufnahme eines Fotos hinzugefügt.

In dem nebenbei laufenden python-skript wird der GPIO abgefrgt. Sobald der Taster betätigt wird simuliert dieser einen Tastendruck (z.B. Strg+P), was den Button betätigt. Gleiches könnte man für alle weiteren Buttons machen. Im Gegensatz zum Beitrag von Martin habe ich den Sleep-Timer etwas erhöht, da mein RPi sich sonst "verschluckt" hat.

Gruß

JS307, 2019-07-23:
Hallo MoJou90, 
ich würde auch Gerne per GPIO auslösen und habe mir die Erklärung von Martin schon angeschaut. Allerdings habe ich keine Erfahrung mit PHP, deshalb scheitere ich schon daran das accesskey attribut in der index.php hinzuzufügen. 
Kannst du mir vielleicht sagen, an welcher Stelle ich hier was einfügen muss, damit ich über die Tastenkombination auslösen kann? 

Grüße & Danke

MoJou90, 2019-07-24:
Moin Moin,
da ich zZ noch immer nicht an mein Image komme, habe ich den part "code" in meinem letzten post noch nicht angepasst.
accesskey ist im weitesten Sinne nur ein Attribut des Buttons
https://wiki.selfhtml.org/wiki/Referenz:HTML/Attribute/accesskey
Zeile 82 müsste damit dann wie folgt aussehen:
`<a href="#" class="btn newpic" accesskey="s"><i class="fa fa-camera"></i> <span data-l10n="newPhoto"></span></a>`

Das python-Skript von "Martin" kann man weitestgehend übernehmen. Allerdings musste für meinen RPI
`time.sleep(0.1)`
auf 
`time.sleep(1)`
angepasst werden.

Auch das Skript werde ich wie versprochen in meinen oberen Post kopieren, sobald ich an mein Image komme.

Gruß
MoJou

MoJou90, 2019-07-30:
Habe nun die betreffenden Änderungen gepostet und in den Code eingefügt. 
Habe Git noch nie verwendet. Vielleicht wird die Änderung ja übernommen.
Gruß
MoJou

andi34, 2019-07-30:
@MoJou90 ich kann's gerne passend in meinen Fork übernehmen und auf Wunsch auch ins Admin Menü packen.

MoJou90, 2019-07-30:
"in meinen Fork übernehmen" heißt in diesem Fall, dass du die Änderung in deine aktuelle Arbeitsversion übernimmst?

andi34, 2019-07-30:
In die Kopie auf meinem GitHub.
Habe das ganze noch etwas erweitert bzw die Erweiterungen anderer mit eingepflegt. Ein Teil ist hier zu sehen https://github.com/andreknieriem/photobooth/pull/58

Ich arbeite im Moment relativ aktiv daran, André hat wohl wenig Zeit im Moment.

JS307, 2019-07-31:
Hallo zusammen, 

vielen Dank für die Unterstützung mit dem accesskey.

Hat denn auch schon mal jemand das eigentliche Thema - 2. Website mit der Galerie und Downloadbutton bearbeitet oder hat vielleicht sogar schon jemand eine Lösung?

obkram, 2019-07-31:
Hallo Andreas, wenn du schreibst „...
In die Kopie auf meinem GitHub ...“ und auf einen Thread verlinkst der eine Version 1.6.2 beinhaltet, heisst das dann das diese Version dann irgendwann auch noch in Andres GitHub zur Verfügung steht oder läuft das jetzt primär über dein GitHub weiter. Wenn das Zweite zutrifft dann wäre eine Verlinkung prima.

andi34, 2019-07-31:
@obkram das ist nicht von mir abhaengig. Ich liebe OpenSource und entsprechend gebe ich auch gerne an das "Mutterprojekt" meinen Code wieder zurueck. Ich denke die meissten werden hier schauen, deshalb schaue ich auch hier vorbei und stelle per pull-request die Aenderungen zur Vefuegung.

@MoJou90 :+1: 
```
From b1801ef8bd2a94a13548ffef019494930f89ce76 Mon Sep 17 00:00:00 2001
From: MoJou90 <52965322+MoJou90@users.noreply.github.com>
Date: Tue, 30 Jul 2019 15:27:52 +0200
Subject: [PATCH] Trigger "takePic" via external button

Add a Python-Script, which reads the state of GPIO24. Pushing a button connected to defined GPIO (GPIO 24) will send a key-combination (alt+s), which triggers the website to take a photo.
To run the python script in background add a cronjob:
sudo crontab -e
@reboot python /home/pi/button.py &

Thanks to Martin ( see comments https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html )
---
 button.py | 20 ++++++++++++++++++++
 index.php |  3 ++-
 2 files changed, 22 insertions(+), 1 deletion(-)
 create mode 100644 button.py

diff --git a/button.py b/button.py
new file mode 100644
index 0000000..5c012d9
--- /dev/null
+++ b/button.py
@@ -0,0 +1,20 @@
+#!/usr/bin/env python
+# Sends alt+s via GPIO24 to take a new picture
+import RPi.GPIO as GPIO
+import time
+import os
+import uinput
+
+#print('Button Pressed')
+
+GPIO.setmode(GPIO.BCM)
+GPIO.setup(24, GPIO.IN, pull_up_down=GPIO.PUD_UP)
+
+while True:
+    input_state = GPIO.input(24)
+    if input_state == False:
+        #print('Button Pressed') #<- for debugging only
+        with uinput.Device([uinput.KEY_LEFTALT, uinput.KEY_S]) as device:
+         time.sleep(1)
+         device.emit_combo([uinput.KEY_LEFTALT, uinput.KEY_S])
+         time.sleep(10)
diff --git a/index.php b/index.php
index ccc6119..bc41a57 100644
--- a/index.php
+++ b/index.php
@@ -63,7 +63,8 @@ require_once('db.php');
 				};
 				?>
 				<?php if($config['use_filter']){ ?><a href="#" class="btn imageFilter"><i class="fa fa-magic"></i> <span data-l10n="selectFilter"></span></a><?php } ?>
-				<a href="#" class="btn takePic"><i class="fa fa-camera"></i> <span data-l10n="takePhoto"></span></a>
+				<!-- accesskey added to take a photo using alt+s (or use an external button) -->
+				<a href="#" accesskey="s" class="btn takePic"><i class="fa fa-camera"></i> <span data-l10n="takePhoto"></span></a>
 			</div>
 		</div>
 
-- 
2.7.4
```

@JS307 am Downloadbutton arbeite ich... leider noch nicht zum laufen bekommen.
http://localhost/images Da sind die Bilder auf jedenfall abrufbar.

andi34, 2019-07-31:
Hardware-Button via admin  panel: https://github.com/andi34/photobooth/commit/fbdf89f9abd28d08ed90973ae207cfd16190a0d5

Nie-Oh, 2019-07-31:
Hallo,

ich möchte mich erst mal bedanken für die tolle Arbeit von euch. Ich bin leider "Endanwender" und kann nicht programmieren. 

Ich möchte trotzdem versuchen einen Beitrag zum Projekt zu leisten.

Mittlerweile habe ich die Photbooth auf einem RPI4 4gb am laufen. Der RPI fuktioniet auch als AP.
Sodass der Extra Router entfallen kann. Natürlich hängt am Rpi ein Touchscreen. Aber auch ein Tablet sollte kein PRoblem sein. 

Ich hatte auch immer das Problem mit der QR-Code Bilder download. Ich habe mich aber an ein ct-Projekt erinnert, die ein RPI als "Litfaßsäule" genutz haben. ich Hoffe ich darf den Link hier Posten.

https://www.heise.de/select/ct/2017/22/1508780300482172

Als Startseite wir die Photobooth angegeben. Egal was , der Client nun aufruft es wird immer die Photobooth angezeigt.

PS: Email-Funktion habe ich bis jetzt noch nicht verwendet.

andi34, 2019-07-31:
Ich nutze den Pi auch als AP. Später wenn das Ding in ne Box kommt wird aber ein Router eingesetzt, da damit das Signal wohl besser zu verteilen ist ;)

Funktioniert auf Pi3 als auch Pi4.

# Set Raspberry as an acess point
````
sudo apt install dnsmasq hostapd
````
Edit /etc/dhcpcd.conf
````
sudo nano /etc/dhcpcd.conf
````
Add to the end of the file
````
interface wlan0
    static ip_address=192.168.4.1/24
    nohook wpa_supplicant
````
restart dhcpcd
````
sudo systemctl restart dhcpcd
````
## Configuring the DHCP server
````
sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf
````
Add the following
````
interface=wlan0      # Use the require wireless interface - usually wlan0
dhcp-range=192.168.4.2,192.168.4.200,255.255.255.0,24h
````
Reload dnsmasq
````
sudo systemctl reload dnsmasq
````
## Configuring the access point host software (hostapd)
````
sudo nano /etc/hostapd/hostapd.conf
````
Add the following
````
interface=wlan0
driver=nl80211
ssid=photobooth
hw_mode=g
channel=7
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=PhotoDownload
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
````
Make sure config gets loaded
````
sudo nano /etc/default/hostapd
````
Uncommend and update DAEMON_CONF
````
DAEMON_CONF="/etc/hostapd/hostapd.conf"
````
Now enable and start hostapd
````
sudo systemctl unmask hostapd
sudo systemctl enable hostapd
sudo systemctl start hostapd
````
Edit /etc/sysctl.conf 
````
sudo nano /etc/sysctl.conf
````
and uncomment this line:
````
net.ipv4.ip_forward=1
````
Reboot once
````
sudo reboot
````
Add a masquerade for outbound traffic on eth0:
````
sudo iptables -t nat -A  POSTROUTING -o eth0 -j MASQUERADE
````
Save the iptables rule.
````
sudo sh -c "iptables-save > /etc/iptables.ipv4.nat"
````
Edit /etc/rc.local 
````
sudo nano /etc/rc.local
````
and add this just above "exit 0" to install these rules on boot.
````
iptables-restore < /etc/iptables.ipv4.nat
````

Source: https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md

tobiashaas, 2019-07-31:
Dem RaspberryPi als Hotspot halte ich für keine gute Idee! 
Verschwendet meiner Meinung nach zu viel Leistung die die Booth gut brauchen kann. 
Stell dir mal vor du hast das Ding auf einer irgendwo stehen. Jeder würde sich damit verbinden, weil jeder denkt - oh gratis Internet 😅. 

Ich nutze dieses kleine Gerät 
GL.iNet GL-AR150-Ext-2 Mini Travel Router with 2dbi external antenna, Wi-Fi Converter, OpenWrt Pre-installed, Repeater Bridge, 150Mbps High Performance, OpenVPN, Programmable IoT Gateway https://www.amazon.de/dp/B01FJ4S9JK/ref=cm_sw_r_cp_api_i_MzyqDbZCP1MEN

Liebe Grüße

MoJou90, 2019-07-31:
Würde den RPi auch eher nicht als AP einsetzen. Aber wenn man grade keinen zur Hand hat ist das sicher eine super Lösung! -> Mit dem RPi 4 ist ja auch noch ein bisschen mehr Arbeitsspeicher verfügbar ;)

Was ich immer machen würde ist es den WLAN-Zugang per QR-Code auszuhängen. Über
https://qifi.org/
kann man dazu recht einfach nen geeigneten QR-Code erstellen.

Nie-Oh, 2019-07-31:
genau so mach ich es auch. Wlan per QR Code verbinden.

Der RPI4 sollt eigentlich genügend leistung haben und es reduziert die Anzahl der Nötigen geräte und Kabelsalt....

Nie-Oh, 2019-07-31:
ich habe die Photobooth schon auf einigen Feiern im Einsatz gehabt. Die Lösung mit dem CT-Projekt fand ich gut, da manche Smartphones prüfen, ob eine Internetverbindung besteht. Wenn nicht, wird das wlan nicht verwendet. Das bedeutet man hat Probleme das Photo runterzuladen. So wie hier schon oft beschrieben. Liegt wohl auch am verwendeten QR Scanner. Deshalb wird ein WLAN mit Internet vorgegaukelt. 

Mit dem "Jeder" verbindet sich, ist nicht so schlimm. Denn Jeder sollte erkennen, wenn das Wlan Photobooth heißt, das es sich um die Photobooth handelt und nicht um einen freien Internetzugang.
Man kann das Wlan auch noch Verschlüsseln. Zugangsdaten stecken dann komplett im QR Code auf der Photobox-Anleitung.

Nie-Oh, 2019-07-31:
Könnte mir bitte jemand eine Version der Photobooth machen, wo der Download Button schon drin ist und die extra Homepage für die Gallery?

Danke

MoJou90, 2019-07-31:
Auch wenn der Titel dieses Threads vermuten lässt, dass es so eine Version schon gibt:
So weit ist es noch nicht ;)

Gruß

obkram, 2019-07-31:
Also meine Erfahrung zum Thema RPI als Access Point ist durchweg positiv. Egal wieviele Leute sich damit verbinden und diverse Bilder parallel herunterladen, das juckt meinen RPI3 überhaupt nicht. Die geringe Reichweite ist sogar ein Vorteil, denn so landen die Gäste schneller wieder im mobilen Datennetz ihres Providers wenn sie sich aus der Reichweite der Box entfernen. Das verhindert das alle grosse Augen bekommen weil sie auf einmal die geladenen Bilder nicht mehr posten können.

ChrisKram, 2019-08-02:
Falls Ihr den Raspi lieber nicht als AccessPoint verwendet, mit einem RPI3 und Touchscreen, hab ich auch massiv Leistungsprobleme, dann bietet sich vllt gerade die Gelegenheit einen MiniTPLInk Router sich anzuschaffen. Hab ich beim letzten BlackWeekend ebenfalls getan. 

https://www.notebooksbilliger.de/blackweekend/tp+link+netzwerk+blackweekend/tp+link+tragbarer+wlan+nano+router+tl+wr802n

Das gleiche Gerät gibt es auch noch mit USB Anschluss für einen 3G/4G Dongle. bzw. gibt es auch für wenig Geld einen TP Link 4G AccessPoint. War für mich interessant, weil ich übelege, die Bilder direkt in die Cloud zuladen und dem QRCode dann den Cloud Link zu geben. So muss ich die User nicht auf die Box selbst lassen, was ich aufgrund der ungeschützen Startseite und Adminpage nicht für praktikabel halte.

F4bsi, 2019-08-05:
Hey @andi34 hast du die Funktion bereits umgesetzt?

Hätte noch ne Idee bzw. einen Funktionierenden Prototypen wie man dies umsetzen könnte, auch ohne eine 2. Website erstellen zu müssen. Meine Idee war das basierend auf einem Cookie einzubauen. Nur wenn dieser durch den Admin Bereich gesetzt wurde wird die Startseite angezeigt.

Also wenn hier noch Unterstützung gebraucht werden kann, würde ich das nochmal raussuchen.

andi34, 2019-08-05:
Habe angefangen aber keine funktionsfähige Lösung. Also gerne wenn du da was hast Mal hochladen :)

Nie-Oh, 2019-08-05:
@andi34 

danke für die schnelle Umsetzung der Greenscreen Funktion.
Für die Gallery hätte ich noch einen Vorschlag wie das funktionieren könte.
Umsetzen muss es ein anderer, da ich vom Programmieren nicht genügend Kenntnis habe.

Mein Vorschlag:
Ruft man die Photobooth mit http://localhost oder http://hostname auf dann wird die Photobooth normal gestartet.

Ruft man aber die Booth von http://ip-Adresse auf wird die Gallerie gestartet.

andi34, 2019-09-05:
WIP: https://github.com/andi34/photobooth/issues/8

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

closed 2019-09-09 by andi34

andi34, 2019-10-11:
Added on 2.0.0- alpha

Open `localhost/gallery.php`

andi34, 2019-10-18:
> Dem RaspberryPi als Hotspot halte ich für keine gute Idee!
> Verschwendet meiner Meinung nach zu viel Leistung die die Booth gut brauchen kann.
> Stell dir mal vor du hast das Ding auf einer irgendwo [...]

 Empfehlungen fürs Setup des Mini Routers ? Habe mir auch Mal einen bestellt, ist angekommen 😅

tobiashaas, 2019-10-18:
Ich habe groß nichts eingestellt. 
Updates gezogen und dem Router gesagt, dass er dem Pi immer die gleiche Ip geben muss. Sonst nichts 😉

Ich bin mittlerweile aber auch nicht mehr so zufrieden mit dem Router .. werde mir den besseren noch besorgen mit 300mbit, der hat nen besseren CPU ;-)