andreknieriem/photobooth #56: Wisch in der Gallery funktioniert nicht direkt auf raspberry pi

koenigd79, 2019-07-12:
Hallo,

habe meine Fotobox nach der aktuellen Anleitung mit dem aktuelsten Raspbian (Buster) neu aufgesetzt. Soweit funzt auch alles.
In der Galerie allerdings kann ich nicht wischen, es werden zum weiterblättern die kleine Pfeile dargestellt.  Ich verwende den Standardbrowser (chromium).
Jemand ne Idee ob ich was für den touch betrieb im Browser einstellen muss??
Beim Zugriff von Tablet funzt es wie es soll.

Kann es am css liegen???

Verwende ein Waveshare 13,3" (https://www.waveshare.com/13.3inch-hdmi-lcd-h-with-case.htm) 

Gruß Daniel

koenigd79, 2019-07-12:
Habs hingekriegt. Scheint ein Problem vom chromium zu sein.
 mit:
@chromium-browser --incognito --kiosk http://localhost/ --touch-events=enabled

funzt es. Das Touchdisplay schein beim Start von chromium nicht richtig erkannt und daher fehlt die Touch unterstützung.

Gruß Daniel

tobiashaas, 2019-07-12:
Vielen Dank Daniel! 
Stand heute Mittag vor dem gleichen Problem... ähnliches Display - welches weiß ich gar nicht genau - wischen ging auch nicht, ebenfalls ging auch der QR Code in der Galerie selbst immer wieder weg wenn man ihn anklickte ;-) Danke dir dafür

closed 2019-08-31 by andreknieriem