andreknieriem/photobooth #57: Admin panel: Hintergrundbild ändern

koenigd79, 2019-07-19:
Toll fände ich die Möglichkeit über das super Admin panel auch noch das Hintergrundbild zu ändern. Zum einen ein pulldown Menü für die schon vorhandenen Bilder und zum anderen noch die Möglichkeit weitere Bilder hochzuladen. So könnte man die Box für jede Feier anpassen, ohne sich tiefer im System anmelden zu müssen.

Gruß Daniel

tobiashaas, 2019-07-19:
Fände ich auch gut, ebenfalls vielleicht auch die Hintergrund Farbe zb. in der Galerie!
Gut fände ich auch wenn man im Admin Panel auch sagen könnte wo die Fotos überall gespeichert werden.

Nur Kamera
Nur Raspi
oder beides

Grüße Tobi

andi34, 2019-07-20:
@koenigd79 glaube nicht so einfach, da es im style.css definiert ist.
https://github.com/andreknieriem/photobooth/blob/master/resources/css/style.css#L264
https://github.com/andreknieriem/photobooth/blob/master/resources/css/style.css#L479

Aber denke es ist nicht schwer einfach selbst das bg.img zu ersetzen wenn gewünscht. Andere Hintergrundbilder gibt es ja auch nicht im Source.

@tobiashaas Instruktionen um das Bild auf der Cam zu behalten ist ja in der README. Muss ja cdm Line angepasst werden und das geht bereits im Admin Panel.
Je nach Cam muss ja ein anderes target definiert sein, also als generelle Option zur Auswahl mMn nicht gut.

tobiashaas, 2019-07-20:
Stimmt, sorry das hab ich übersehen, Danke dir

andi34, 2019-07-20:
Die cmd Line wäre dann nicht per hand änderbar. Oder du bräuchtest eine Checkbox zusätzlich.
Checkbox: Cmd Line per Hand aktivieren = true 
--> Auswahl aus Liste hat keine Funktion.
Checkbox: Cmd Line per Hand aktivieren = false
--> Auswahl aus Liste aktiv
Nur Kamera:  macht keinen Sinn da Galerie dann überflüssig und kein Bild auf Bildschirm
Nur Pi: $config['take_picture']['cmd'] = 'sudo gphoto2 --capture-image-and-download --filename=%s images';
Pi und Cam1: $config['take_picture']['cmd'] = 'sudo gphoto2 --capture-image-and-download --keep --filename=%s images';
Pi und Cam2: $config['take_picture']['cmd'] = 'sudo gphoto2 --set-config capturetarget=0 --capture-image-and-download --keep --filename=%s images';
Pi und Cam3: $config['take_picture']['cmd'] = 'sudo gphoto2 --set-config capturetarget=1 --capture-image-and-download --keep --filename=%s images';

Ob 1, 2 oder 3 würde man nur durch probieren erfahren. Da ist es einfacher die Instruktionen im README zu befolgen. Jede Cam hat im Zweifel ein anderes target.

Korrigiert mich wenn ich falsch liege.

tobiashaas, 2019-07-20:
Meine Überlegung war dass man 2 Checkboxes anlegt und die CMD Line komplett rausnimmt.
So dass man nur noch anwählen muss was man haben möchte.
- save only on Pi
- save on Pi & Camera

Aber wenn das CaptureTarget je nach Kamerahersteller unterschiedlich ist, macht das kein Sinn, da hast du Recht! ;-)

andi34, 2019-07-20:
Ja, gibt cams mit internem Speicher soweit ich weiss. Auch ist capturetarget glaub nicht bei allen cams unterstützt.

koenigd79, 2019-07-20:
Ja so mache ich da aktuell auch, einfach die bg.jpg entsprechend anzupassen.

Dachte halte wäre ne feine sache, das über das Admin panel zu erledigen. Aber klar liegt im ccs und nicht in den php files, das lässt sich dann nicht so einfach realisieren....

Ansonsten sind deine weiterentwicklungen super andi. Habe mir vom https://www.reinhard-rebernig.at/website/websites/fotokasterl/ schicken lassen. Schau mal ob ich die filter ans laufen kriege.

andi34, 2019-07-20:
@koenigd79 hab ich schon am laufen. Sind nur noch Feinheiten. Denke nächste , spätestens übernächste Woche lade ich es hoch für alle

andi34, 2019-07-20:
Aber wenn du Lust hast: Lad doch dein bg.img Mal hoch. Dann können auch andere es nutzen als Alternative.

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

closed 2019-09-09 by andi34