andreknieriem/photobooth #64: Druckproblem und QR Code

danielovic295, 2019-09-01:
Hallo,

meine Bilder werden nicht gedruckt, jedoch im Verzeichnis print abgelegt. Anstatt zu drucken wird die Kamera ausgelöst. Drucker läuft via CUPS und kann Testseiten drücken. 
Außerdem hätte ich gern gewusst, wie der QR Code nicht mit ausgedruckt wird. 

Vielen Dank für die Hilfe und Respekt für das tolle Projekt. 

Viele Grüße 
Daniel

andi34, 2019-09-02:
Schau Mal hier: https://github.com/andreknieriem/photobooth/commit/5c379174d215e913d12ba226e76384b7bd3867be

Gab vorher den Fehler, dass er den Take pic command genommen hat, wenn man per Admin Panel die config gespeichert hat.
Einfach Mal überprüfen was bei die als Print cmd steht.

ChrisKram, 2019-09-03:
Ich würde mich der Frage gern anschließen, ob schon implementiert ist, dass der QR Code nicht mit gedruckt wird. 

Der Nutzen des QR Codes auf dem Foto erschließt sich mir nicht, da es ja kein Online Code ist.

andi34, 2019-09-03:
Versuch Mal diiese Line auszukommentieren:
https://github.com/andreknieriem/photobooth/blob/master/print.php#L51

Also
```
// imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));
``` 
Könnte ausreichen auf die schnelle betrachtet

ChrisKram, 2019-09-03:
Danke für die schnelle Antwort. Den Vorschlag hab ich noch nicht ausprobiert. Habe zwei Varianten getestet: 
Alles zwischen ersten if  und dem print.cmd auskommentieren. Dann wird direkt gedruckt, aber kein Bild zwischen gespeichert

Und zweitens, ich hab die QR Code Erstellung und weitere Teile davon auskommentiert. So wird ein Bild im Print Ordner abgelegt. Ist zumindest für Testzwecke einfacher, weil man das Ergebnis sieht ohne zu Drucken. Die Druckaufträge selbst, kann man sich in der Queue von CUPS ansehen. 
Sieht dann so aus.

```
// print
if(file_exists($filename_source)) {
    // copy and merge
    if(!file_exists($filename_print)) {
        // create qr code
        // if(!file_exists($filename_codes)) {
            // include('resources/lib/phpqrcode/qrlib.php');
            // $url = 'http://'.$_SERVER['HTTP_HOST'].'/download.php?image=';
            // QRcode::png($url.$filename, $filename_codes, QR_ECLEVEL_H, 10);
        // }

        // merge source and code
        list($width, $height) = getimagesize($filename_source);
        // $newwidth = $width + ($height / 2);
        // $newheight = $height;

        $source = imagecreatefromjpeg($filename_source);
        // $code = imagecreatefrompng($filename_codes);
        $print = imagecreatetruecolor($width, $height);

        // imagefill($print, 0, 0, imagecolorallocate($print, 255, 255, 255));
        imagecopy($print, $source , 0, 0, 0, 0, $width, $height);
        // imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));

        imagejpeg($print, $filename_print);
        imagedestroy($print);
        //imagedestroy($code);
        imagedestroy($source);
    }

    // print image
    // fixme: move the command to the config.inc.php
    $printimage = shell_exec(
        sprintf(
            $config['print']['cmd'],
            $filename_print
        )
    );
    echo json_encode(array('status' => 'ok', 'msg' => $printimage || ''));
}
```

danielovic295, 2019-09-05:
Das Druckproblem ist leider noch nicht behoben, die index.php ist mittlerweile in der aktuellsten Version, die vorgeschlagene Lösung bringt mich also nicht weiter.

In der GUI von CUPS kommen keine Druckaufträge an, wenn ich via Photobooth drucke. Via Kommandozeile funktioniert das Drucken aber. Ich verwende jeweils den selben Befehl. 

Hat jemand eine Idee?

ChrisKram, 2019-09-05:
Klären wir doch mal die Basics, da ich gerade erst vorgestern die neueste Version mit Printing eingerichtet habe, weiss ich, dass es funktionieren kann.

Welchen Drucker hast du?
Ist er als Standarddrucker in Cups festgelegt?
Welche Version der Photobooth verwendest du? 
Wird immer noch die Kamera ausgelöst?

danielovic295, 2019-09-05:
Ich benutze einen DNP DN RX-1HS via Gutenprint 5.2.11. Er ist als Standarddrucker auch so festgelegt, er ist auch der einzige Drucker, was CUPS lt. eigenem  manual sowieso dazu veranlasst, ihn als Standarddrucker zu behandeln.

Ich benutze das aktuellste Release der Photobooth (1.5.2). Die Kamera wird, seit ich die neuste Version verwende, auch nicht mehr ausgelöst.

Der Druckbefehl der Photobooth funktioniert via Kommandozeile problemlos.

ChrisKram, 2019-09-05:
Tatsächlich ist das aktuelle Release die Version 1.8.x, am besten noch mal auf die aktuelle Version aktualisieren und die readme.md hier in github unter "code" lesen. Dort kannst du auch den aktuellen code runterladen (rechts open unter "Clone or Download").

Unter Linux kannst du dir das aktuelle Paket auch mit 
`sudo git clone https://github.com/andreknieriem/photobooth html`
holen.

Benenne vorher den html Ordner um und lege einen neuen an. Dann kannst du jeder Zeit auf die andere Version zurück. Gib dem neuen html folder wieder entsprechend Rechte (siehe readme).

danielovic295, 2019-09-05:
Habe mir den Code gestern erst via git clone geholt, deshalb habe ich wohl die aktuellste Version. Bin bei meiner Aussage bzgl. der Version von der neusten Release Version ausgegangen, die hier auf github released wurde (Im Abschnitt "Releases") 

Meinen raspberry habe ich schon komplett neu aufgesetzt, da ich da viel herum experimentiert und daher Probleme vermutet habe. Tatsächlich hat die Neuaufspielung einige Bugs behoben, leider hat es mich in Sachen drucken nicht weiter gebracht.

Ich werde nun nochmals einen Versuch starten und Deinem Rat folgen, die Software lt. README nochmals neu aufspielen. 
Ich bin zwar in Sachen Linux wirklich Neuling, logisch ist meine Problematik für mich trotzdem nicht. In der Kommandozeile funktioniert der Befehl, aber der Ausdruck via Photobooth läuft nicht. 

Trotzdem vielen Dank für Deine Mühen, mir hier weiter zu helfen.

devnull1975, 2019-09-05:
Bei mir läuft mein Selphy auch nur wenn ich ihn via USB am Pi anschließe. Per WLan, der Pi ist hier Hals AP eingerichtet,  geht's bei mir auch nicht. 
Wie ist dein Drucker angeschlossen?

ChrisKram, 2019-09-05:
Du kannst dir auch erstmal die Logfiles des Apache ansehen und schauen, ob da eine brauchbare Information drin steht.

Letztlich wird ja durch die Webseite ein Terminalbefehl ausgeführt, daher kann der einfachste Fehler schon sein, dass die Anwendung dazu keine Berechtigung hat. Das würde auch erklären, warum die Kamera nicht ausgelöst wird. 

Bevor du alles neu machst, fang erstmal damit an.

danielovic295, 2019-09-05:
Mein Drucker ist via USB am Pi angeschlossen, der Pi fungiert aber auch als AP für das Tablet.

Die Kamera löst aus, speichert das Bild in html/images/. Wenn ich das Bild via Photobooth drucken möchte, wird das Bild auch in /html/print abgelegt. Die Bilder aus dem print Verzeichnis lassen sich problemlos via Kommandozeile ausdrucken.

Jedenfalls hat mich der Tipp mit der apache logfile ans Ziel geführt. Dort wurde in der error.log die Meldung "lp: Not allowed to print." protokolliert. Nach kurzer Googelei kann ich nun wie folgt drucken:

sudo lp -U pi s%

Linux, sein Berechtigungssystem und ich werden wohl keine Freunde mehr.. 

Vielen, vielen Dank für Eure Unterstützung, man kann immer nur dazu lernen. 

Viele Grüße
Daniel

andi34, 2019-09-05:
Problem gelöst,
QR optional eigenes Issue jetzt: https://github.com/andreknieriem/photobooth/issues/66

closed 2019-09-05 by andi34