andreknieriem/photobooth #65: Startseite abgeschnitten

SmilingJack85, 2019-09-02:
Hi,
habe jetzt auch endlich die Fotobox am laufen, funktioniert auch alles wunderbar.
Als Display nutze ich einen 7 Zoll Touchscreen am Pi, mit 1024x600. 
Leider ist der Button für die Filterwahl kaum zu sehen. Aber nur auf dem Pi Display, 
Wo kann ich das denn anpassen, bzw wo könnte mein Fehler liegen? Ich bekomm das nicht hin 😒

andi34, 2019-09-02:
Desktop Seite Verwenden. Design ist nicht responsive.
Gibt auch im Admin panel die option die desktop Seite zu erzwingen.

https://github.com/andreknieriem/photobooth/commit/d5152aa83eb3aa574020b16e532ef189ac952411
https://github.com/andreknieriem/photobooth/commit/18919f85ea5a479e12137a38ab8f0333f96b03de

andi34, 2019-09-02:
https://github.com/andreknieriem/photobooth/issues/22

closed 2019-09-02 by andi34