andreknieriem/photobooth #66: Make QR Code optional on print

ChrisKram, 2019-09-05:
Im Moment kann ein Bild per Default nur mit QR Code gedruckt werden. Dafür wird das Bild kleiner dargestellt und der QR Code rechts neben das Foto gesetzt. 

Die ist im Moment, da der Barcode keinen externen Link in die Cloud enthält nur während der Veranstaltung verwendbar und somit eher schade um das Bild. Daher wäre es praktisch, den Barcode optional zu gestalten und per Admin Panel für den Print ein oder ausschaltbar zu machen.

_In english please:_
_Actually the qr codes is printed onto the foto by default. A resizing happens and the qr code is added to the left.

Since the barcode is only useful while the party happens it would be great when the barcode is optional and an additional option in the admin panel allows it to deactivate it._

andi34, 2019-09-05:
Wie bereits im anderen Issue schon erwähnt (keine Rückmeldung erhalten):
Versuch Mal diiese Line auszukommentieren:
https://github.com/andreknieriem/photobooth/blob/master/print.php#L51

Also
```
// imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));
``` 
Könnte ausreichen auf die schnelle betrachtet.

Ist kein Thema das dann optional zu machen wenn's reicht, bekomme ich auch kurzfristig am Wochenende ins Admin Menü gepackt.

ChrisKram, 2019-09-05:
Ich weiß noch nicht ob ich es heute oder morgen noch schaffe, brauche die Box am WE und muss noch andere Dinge ändern, bevor es losgeht.

flighter18, 2019-09-05:
Hi @ChrisKram 

Ich wollte auch, dass nur das Bild ohne QR Code gedruckt wird. 
Um dies zu erreichen, musste ich diverse Zeilen in der print.php Datei auskommentieren. Zusätzlich, musste ich eine Zeile hinzufügen.

Im Anhang meine print.php Datei. Vielleicht hilft Dir das weiter.
Bitte beachte, ich musste die Dateiendung von php auf txt ändern. 

[print.txt](https://github.com/andreknieriem/photobooth/files/3581269/print.txt)

ChrisKram, 2019-09-05:
> 
> 
> 
> ```
> // imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));
> ```
> 
> Könnte ausreichen auf die schnelle betrachtet.

@andi34 das reicht leider nicht aus. Dann ist zwar der QR Code nicht mit auf dem Bild, es wird aber weiterhin verkleinert. Ich denke, dass es dafür eine else Bedingung braucht in der dann nur der print job aufgerufen wird.

Ungefähr so (habs nicht ausprobiert und bin nicht vertraut mit php - bitte noch mal prüfen) - siehe Anhang.
[print.txt](https://github.com/andreknieriem/photobooth/files/3581555/print.txt)

@flighter18 - danke für den Vorschlag. ich habe mir eine ähnliche Version zusammen gebaut. Siehe #64

andi34, 2019-09-06:
Danke @ChrisKram und @flighter18 , ich schau Sonntag nochmal rein

andi34, 2019-09-07:
@ChrisKram falls du testen willst
```
cd /var/www/html
git config core.fileMode false
git add --all && git commit -a -m "Squash"
git remote add andi34 https://github.com/andi34/photobooth
git fetch andi34 dev
git checkout andi34/dev
sudo mkdir -p /var/www/html/images
sudo mkdir -p /var/www/html/keying
sudo mkdir -p /var/www/html/print
sudo mkdir -p /var/www/html/qrcodes
sudo mkdir -p /var/www/html/thumbs
sudo mkdir -p /var/www/html/tmp
sudo chown -R pi: /var/www/
sudo chmod -R 777 /var/www
```

andi34, 2019-09-08:
Works fine.
Merged.

closed 2019-09-08 by andi34