andreknieriem/photobooth #67: Eigene Rahmen, Text oder Logos für den Fotodruck

ChrisKram, 2019-09-05:
Neben dem barcode wäre eine Möglichkeit eigene Daten oder Rahmen mit zu drucken eine tolle Option. 

So kann bei einer Hochzeit bspw. das Datum, die Namen, Ort, etc. mit gedruckt werden. Weiterhin könne man einen eigenen Rahmen, der die Veranstaltung beschreibt oder einen Anlass, z.B. ein Mistelzweig zu Weihnachten ect. erlauben oder Logos mit Drucken. 

Ein Rahmen kann auch ein Hintergrund sein, der unter das Foto gelegt wird und so das skalierte Bild umschließt. 

Ein Chroma Keying ist dafür nicht zwingend erforderlich.

andi34, 2019-09-05:
On my list:
https://github.com/andi34/photobooth/issues/22

andi34, 2019-09-05:
https://github.com/andreknieriem/photobooth/issues/9

andi34, 2019-09-08:
[image]

andi34, 2019-09-08:
frame.png needs to be added to root of photobooth.
https://github.com/andi34/photobooth/commit/20a9b16623431942aed1610e843ae6f2e665366b#diff-d5fde551bb3c77d93baa17a8ba21e781R58

If someone can share a frame which can be used in this project without license issues let me know.
No frame.png added by default until that.

andi34, 2019-09-08:
added example frame

[image]

tobiashaas, 2019-09-08:
Ich hab dir ne Mail mit einem Frame geschickt!

andi34, 2019-09-08:
Danke :)
Hast schon getestet damit?
Bin grad nur mobil, müsste heute Abend am PC dann schauen.

tobiashaas, 2019-09-08:
Leider noch nicht... Hab grade meinen Raspi nicht zur Hand ...

andi34, 2019-09-08:
Ok, links und rechts ist noch Teil vom Bild Eigentlich, somit sollte dein Rahmen nicht abschließen am Rand . Muss Mal schauen wie ich das behoben bekomme dann.
Aber bekommen wir hin :)

andi34, 2019-09-09:
discussion/development continues here https://github.com/andreknieriem/photobooth/issues/68

closed 2019-09-09 by andi34