andreknieriem/photobooth #68: Feature request collection

andi34, 2019-09-09:
**Ready for review**
- [X] frame around the picture for printing https://github.com/andi34/photobooth/commit/ccc6333a55374cd9b9e5ac2df770a5e7939a3541

**Work in progress (WIP)**
- [ ] text overlay for photo printing
- [ ] print confirmation

**Help wanted**
- [ ] download from gallery instead QR Code

thymon13, 2019-09-09:
## Bouton on gallery  <h2>

The buttons on the photos from the gallery are too small and especially unsightly.

[image]

Would it be possible to enlarge them?
It would be best to have them in the middle of the picture. 
CSS buttons that appear and disappears when click on the photo. ?

like this for exemple :
[image]
Format: [image]

thymon13, 2019-09-09:
## QR Code Pop'up not resized <h2>

I use a 9.6" touche pad. And the QR code is truncated.
I'have try with "false" ou "true" in my config for enable or not the mobile view. 
No change.

[image]

andi34, 2019-09-09:
> ## QR Code Pop'up not resized
> 
> I use a 9.6" touche pad. And the QR code is truncated.
> I'have try with "false" ou "true" in my config for enable or not the mobile view.
> No change.
> 
> [image]
@thymon13 https://github.com/andreknieriem/photobooth/issues/22

> ## Bouton on gallery
> 
> The buttons on the photos from the gallery are too small and especially unsightly.
> 
> [image]
> 
> Would it be possible to enlarge them?
> It would be best to have them in the middle of the picture.
> CSS buttons that appear and disappears when click on the photo. ?
> 
> like this for exemple :
> [image]

This is also a design thing and not a feature itself.

thymon13, 2019-09-09:
## Chroma Keying <h2>

I tested this function. Thats work perfectly.
But is it possible to use a picture background by default ? Instead of choose them in gallery -> select photo -> select function --> select background...

andi34, 2019-09-09:
> ## Chroma Keying
> I tested this function. Thats work perfectly.
> But is it possible to use one picture background by default ? Instead of choose them in gallery -> select photo -> select function --> select background...
> 
> Example : Corporation event with Enterprise Logo in background by default...

You can adjust the source for your needs https://github.com/andreknieriem/photobooth/blob/master/chroma_keying/index.php#L51

thymon13, 2019-09-09:
> This is also a design thing and not a feature itself.
@andi34 really ? :sob: :sob:

thymon13, 2019-09-09:
I found anothers error for french translation...

thymon13, 2019-09-09:
>you can adjust the source for your needs 
https://github.com/andreknieriem/photobooth/issues/68#issuecomment-529620150

I think my question is not clear.  When you take a picture the result is with the green background.  and you have to go to the gallery to choose the background you want to incorporate.  While it would be nice to have the result directly when the picture is taken.

andi34, 2019-09-09:
That's nothing I am going to make an option. You can take a look at the chroma_keying source and modify the takePic.php for it, or use the new frame function (can also be added to takePic.php to apply the frame directly).

andi34, 2019-09-09:
> > This is also a design thing and not a feature itself.
> 
> @andi34 really ?  
> So I open a design issue ?
Could also be part of the responsive design.
I am not able to implement every feature request myself (I've a full time job as a medical assistant and still learning js, php etc). Everyone is welcome to adjust the source for his/her needs and send a Pullrequest with useful changes (sure, I can make it clean commits in the end as I have a lot of experience with git).

Going to write a quick "how to submit code changes" soon.

andi34, 2019-09-09:
> I found anothers errors for french translation...

Fork latest source, update the language file and send me a link for it. Maybe there's more to be adjusted?

thymon13, 2019-09-09:
Ok thank you Andi34 for your answer.

andi34, 2019-09-09:
@thymon13 maybe useful for you until ready:
```
git fetch https://github.com/andi34/photobooth private
git cherry-pick d317f58d95752ac585c8e2545ef24957ceb11a4c
git cherry-pick 9dab3dfae43342952ba5f6db85c5123aa3869c0b
git cherry-pick 250172ae175e6f815375335dcce566b840ad7bb2
```

thymon13, 2019-09-09:
news private fetch for what ?

and how you commit ?
```
git cherry-pick d317f58d95752ac585c8e2545ef24957ceb11a4c
error: Your local changes to the following files would be overwritten by merge:
	resources/css/style.css
	resources/js/core.js
	resources/sass/style.scss
Please commit your changes or stash them before you merge.
Aborting
fatal: cherry-pick failed
```

andi34, 2019-09-09:
This would create a commit with your made changes, yes. After that you can cherry-pick

thymon13, 2019-09-09:
@andi34 
I'm really sorry but I do not understand anything

```
git cherry-pick d317f58d95752ac585c8e2545ef24957ceb11a4c
error: your local changes would be overwritten by cherry-pick.
hint: commit your changes or stash them to proceed.
fatal: cherry-pick failed
```

andi34, 2019-09-10:
```
git cherry-pick --abort
git add --all
git commit -a -m "My local changes"
```
After that it should work. You'll only have conflicts at cherry-pick if you've modified the source at the same places already.

thymon13, 2019-09-10:
@andi34  ok it's done

andi34, 2019-09-10:
Maybe the adjustments for smaller device helps you a little bit

thymon13, 2019-09-10:
Well no... :cry:

look : https://youtu.be/J8yeS-4Nu7s

andi34, 2019-09-11:
What's your resolution?
You could try to adjust the ```@media (max-width: 800px)``` inside the style.css

thymon13, 2019-09-11:
1280x800px
I will test it, thx

vdubuk, 2019-09-11:
Hi @andi34 
Text on print, I do not know how to combine into repository, bur here is the code ive used 

print.php   after  $status = false;
`
// text on print variables -start
$fontsize = $config['fontsize']; 

$fontlocx = $config['locationx'];
$fontlocy = $config['locationy'];
$linespacing = $config['linespace'];
$fontrot = $config['rotation'];
$line1text = $config['textonprint']['line1'];
$line2text = $config['textonprint']['line2'];
$line3text = $config['textonprint']['line3'];
// text on print variables -end
`
after   imagecopyresized($print, $code, $width, 0, 0, 0, ($height / 2), ($height / 2), imagesx($code), imagesy($code));
`           // text on image - start
            if($config['is_textonprint'] == true) {
            $fontcolour = imagecolorallocate($print, 0, 0, 0);  // colour of font
            imagettftext ($print, $fontsize, $fontrot, $fontlocx, $fontlocy, $fontcolour, 'resources/fonts/GreatVibes-Regular.ttf' , $line1text);
            imagettftext ($print, $fontsize, $fontrot, $fontlocx, $fontlocy + $linespacing, $fontcolour, 'resources/fonts/GreatVibes-Regular.ttf' , $line2text);
            imagettftext ($print, $fontsize, $fontrot, $fontlocx, $fontlocy + ($linespacing *2), $fontcolour, 'resources/fonts/GreatVibes-Regular.ttf' , $line3text);
            }
            //text on image - end
`
in admin/index.php add
`// text print start
		'textonprint' => [
			'is_textonprint' => [
				'type' => 'checkbox',
				'name' => 'is_textonprint',
				'value' => $config['is_textonprint']
			],
			'line1' => [
				'type' => 'input',
				'placeholder' => 'line 1',
				'name' => 'textonprint[line1]',
				'value' => $config['textonprint']['line1']
			],
			'line2' => [
				'type' => 'input',
				'placeholder' => 'line 2',
				'name' => 'textonprint[line2]',
				'value' => $config['textonprint']['line2']
			],
			'line3' => [
				'type' => 'input',
				'placeholder' => 'line 3',
				'name' => 'textonprint[line3]',
				'value' => $config['textonprint']['line3']
			],
			'locationx' => [
				'type' => 'input',
				'placeholder' => 'locationx',
				'name' => 'locationx',
				'value' => $config['locationx']
			],
			'locationy' => [
				'type' => 'input',
				'placeholder' => 'locationy',
				'name' => 'locationy',
				'value' => $config['locationy']
			],
			'rotation' => [
				'type' => 'input',
				'placeholder' => 'rotation',
				'name' => 'rotation',
				'value' => $config['rotation']
			],
			'fontsize' => [
				'type' => 'input',
				'placeholder' => 'fontsize',
				'name' => 'fontsize',
				'value' => $config['fontsize']
			],
			'linespace' => [
				'type' => 'input',
				'placeholder' => 'linespace',
				'name' => 'linespace',
				'value' => $config['linespace']
			],
		],
		// text print end
`
in lang/en.js  (sorry ive not been able to add German or French) add

"textonprint": "Add Text to Print",
	"is_textonprint": "Text on print",
	"textonprint_line1": "Text for line 1",
	"textonprint_line2": "Text for line 2",
	"textonprint_line3": "Text for line 3",
	"textonprint_locationx": "X Coordinate",
	"textonprint_locationy": "Y Coordinate",
	"textonprint_rotation": "Text rotation",
	"textonprint_linespace": "Line spacing",
	"textonprint_fontsize": "Font size"

andi34, 2019-09-11:
@vdubuk great!
You're on latest 1.8.1 branch?
You've a clone on your git?

Easiest might be:
```
git config core.fileMode false
git add --all
git commit -a -m "Add print text on pictures"
git format-patch-1
```
Now open the generated patch file and create a new gist here: https://gist.github.com/ and add the content of the patch file there and share the link to the gist.

vdubuk, 2019-09-11:
> You're on latest 1.8.1 branch?

erm.. how do i check branch.

I made a clean install of andreknieriem/photobooth this morning

andi34, 2019-09-11:
```
git log --oneline
```
This will show latest commits on your branch. If you clean cloned this morning you are and only need to run mentioned commands in terminal (cd to photobooth source first).

andi34, 2019-09-11:
After the ```git commit -a -m "Bla"``` you can also do this:
``` git checkout -b printtext
git push https://github.com/vdubuk/photobooth HEAD: printtext```
This will create a new branch with your changes included on your clone.

thymon13, 2019-09-11:
> What's your resolution?
> You could try to adjust the `@media (max-width: 800px)` inside the style.css

Forget about it, I found it why.
I did not want to see the web browser address bar. I have install an app for that.
Because when I test with Google Chrome, works.

sorry for any wasted time

andi34, 2019-09-11:
> @andi34
> 
> > What's your resolution?
> > You could try to adjust the `@media (max-width: 800px)` inside the style.css
> 
> Forget about it, I found it why.
> I did not want to see the web browser address bar. I have install an app for that.
> Because when I test with Google Chrome, works.
> 
> sorry for any wasted time

Start in kiosk mode?
https://github.com/andi34/photobooth#kiosk-mode

thymon13, 2019-09-11:
@andi34 
I can't, i use a Android Tablet...
I am looking for how to modify the default web browser to use Google chrome

andi34, 2019-09-13:
@vdubuk thanks a lot! i've done some small adjustments.

@thymon13 could you do the french and spanish translation and post them for me?
i'll take care about german later.

https://github.com/andi34/photobooth/commit/742320da84c0c73ac16461e5c0175a4e9029aa72

andi34, 2019-09-13:
- [ ]  initial theme support (switch between default and bluegrey theme via config) https://github.com/andi34/photobooth/commit/d58094697d60a89bf8b0800bd74aec11686cbb3e (translation need to be done)

thymon13, 2019-09-14:
@andi34 
French translation it is done.
Spanish translation : 
I'm waiting for my wife because she's a Spanish teacher 🤣

andi34, 2019-09-15:
Thanks @thymon13 , applied: https://github.com/andi34/photobooth/commit/a6502618fd539ce478571690251565fe83f5de90

Could you also check the theme support translation?

vdubuk, 2019-09-15:
would it be possible to use PiCamera for Preview and Photo ?

looking at apache error logs it seems that when using the PiCamera for Preview it cannot be accessed in the command line to take Photo and therefore and error happens. 

Would it be possible to stop the Preview just before the Photo function to take the photo is called?

andi34, 2019-09-15:
Should be possible to stop the stream once countdown is called, but not sure how to stop the stream

thymon13, 2019-09-15:
@andi34 
all is done now :)

andi34, 2019-09-15:
No problem and thanks :)
I'll take care of it tomorrow if I am up early enough :)

andi34, 2019-09-16:
Thanks @vdubuk for your implementation.
Thanks @thymon13 for the translation :)

- [X] text overlay for photo printing ( @vdubuk , v1.8.2 )
- [X] initial theme support: switch between default and bluegrey theme via config (v1.8.2 )

Both merged :)

msmedien, 2019-09-16:
Besteht die Möglichkeit ein fest einprogrammiertes "Overlay" für den/die Drucken-Buttons zu implementieren?

Immer wenn die Gäste etwas drucken möchten, wird zum Teil mehrfach gedrückt, da auf dem Bildschirm nichts angezeigt wird. Ich habe daher die angehangene Lösung gefunden, welche sehr gut funktioniert und in der Struktur schön einfach ist. - Das Problem ist halt nur das ich es bei jedem Update neu einbinden muss. Dabei würde es sicherlich mehreren Leuten helfen. (Bzw. den Gästen)

Als Erweiterung wäre es cool wenn man den angezeigten Text in der Sprachdatei anpassen könnte. :)

[Druckbenachrichtung.txt](https://github.com/andreknieriem/photobooth/files/3615925/Druckbenachrichtung.txt)

flighter18, 2019-09-16:
Hi @msmedien 
Ein Lösung fur dein "Problem" ist in arbeit.  Dazu gibt es auch ein "To Do". Du findest es unter folgendem Link findest:
https://github.com/andi34/photobooth/issues/20#issuecomment-531187459

Hier noch das Beispiel Video für die implementierte Lösung.
https://youtu.be/NTxXxTOfN8U

Möchtest du die Implementation jetzt schon haben, kannst du dieses Repo Clonen:
https://github.com/flighter18/photobooth/tree/printconfirm

### ACHTUNG Branch heisst printconfirm

andi34, 2019-09-16:
> Besteht die Möglichkeit ein fest einprogrammiertes "Overlay" für den/die Drucken-Buttons zu implementieren?
>
> Immer wenn die Gäste etwas drucken möchten, wird zum Teil mehrfach gedrückt, da auf dem [...]

Danke für die Anregung. Ich habe noch nicht geschafft neben Vollzeitjob und Umzugsstress den Code selbst zu testen und anzupassen auf die einzelnen Sprachen, ist aber auf meiner Liste und spätestens nächste Woche sollte ich Luft dafür haben.

Wenn du dich mit git etwas auskennst einfach cherry-picken, geht auch automatisiert per Script (gerne hänge ich dir heute Abend mein Script Mal an).

andi34, 2019-09-16:
```
cd var/www/html
git fetch https://github.com/flighter18/photobooth printconfirm
git cherry-pick 22ae559d63cc2bc737449062b0aeb0246080f41a
git cherry-pick e41a760cd362f9de00d7eb6dcee54c5814f89d02
```
Nur drauf achten das du die Filemode Änderungen vorher ausgestellt hast im repo.
```
cd /var/www/html
git config core.fileMode false
sudo git config core.fileMode false
```

andi34, 2019-09-16:
@msmedien gerade erst gerallt das du auch eine Implementierung hast 🤗
Diese Teste ich natürlich auch, finde die Idee gut auf externe Scripts zu verzichten wie es bei dir der Fall ist.

andi34, 2019-09-16:
@msmedien I'll pushed your changes here for now. I'll make it one clean commit and do some adjustments within this week. https://github.com/andi34/photobooth/commits/dev

andi34, 2019-09-16:
**Ready for review**
- [ ] print confirmation ( @flighter18 , @msmedien ): 
  https://github.com/andi34/photobooth/commit/3272cb7b96948d5aa9c6f6c353f392ddeebd87a2
  https://github.com/andi34/photobooth/commit/bc3daf1c571124bf483b9b84dbb13b6a7a2a5de9

andi34, 2019-09-16:
> would it be possible to use PiCamera for Preview and Photo ?
>
> looking at apache error logs it seems that when using the PiCamera for Preview it cannot be accessed in the command line to take Photo [...]

Could you try to move this parts up a little bit?
https://github.com/andreknieriem/photobooth/blob/84b3c21647376f01e73c436682cd8466975979b0/resources/js/core.js#L123-L127

So it looks like:

    // take Picture
    public.takePic = function (photoStyle) {
        processing = true;
        if (isdev) {
            console.log('Take Picture:' + photoStyle);
        }
        if(useVideo){
            var track = public.stream.getTracks()[0];
            track.stop();
            $('video').hide();
        }
        setTimeout(function () {
	    if ((photoStyle=='photo')){
                $('#counter').text('');
                $('.spinner').show();
                $('.loading').text(L10N.busy);
            } else {
```

thymon13, 2019-09-17:
@andi34 
Nice Tutorial ! Need to add this en (FAQ)

I tested "Print Confirmation" Soooooo great ! Works perfectly

andi34, 2019-09-18:
**Feature added**
- [X] print confirmation ( @flighter18 , @msmedien )

Merged :smiley: 
https://github.com/andreknieriem/photobooth/commit/ab2ebf3f551d194187b9918a926ad75aa70468fd
https://github.com/andreknieriem/photobooth/commit/58fb383570e57acaa654e5c4a8b052d21efeb7d6

Thanks everyone :+1:

andi34, 2019-09-18:
**Ready for review**
- [ ] polaroid effect: https://github.com/andi34/photobooth/commit/df3b84dcabdeeb3c0a05d8d4f944221ecc2da1bd

Might need adjustment to also use $filename_orig = $filename_tmp;
 
change from
```
if($_POST['style'] === 'photo') {
	$filename_orig = $filename_photo;
	$use_filter = false;
} else {
	$filename_orig = $filename_tmp;
	$use_filter = false;
}
if(!empty($_POST) && !($_POST['filter'] === 'imgPlain')) {
	$filename_orig = $filename_tmp;
	$use_filter = true;
	$imgfilter = $_POST['filter'];
}
```
to
```
if($_POST['style'] === 'photo') {
	$filename_orig = $filename_photo;
	$use_filter = false;
} else {
	$filename_orig = $filename_tmp;
	$use_filter = false;
}

if($config['polaroid_effect'] === true) {
	$filename_orig = $filename_tmp;
}

if(!empty($_POST) && !($_POST['filter'] === 'imgPlain')) {
	$filename_orig = $filename_tmp;
	$use_filter = true;
	$imgfilter = $_POST['filter'];
}
```

Examples:
[image]

[image]

andi34, 2019-09-18:
Btw. You can follow the steps mentioned here again to test this new feature:
https://github.com/andreknieriem/photobooth/issues/68#issuecomment-531877199

If you've already added my fork as remote you only need to run
```
cd /var/www/HTML
git fetch andi34
git checkout andi34/dev
```

vdubuk, 2019-09-19:
> Could you try to move this parts up a little bit?
> https://github.com/andreknieriem/photobooth/blob/84b3c21647376f01e73c436682cd8466975979b0/resources/js/core.js#L123-L127
>
> So it looks like:
>
> ``` [...]

This solution worked for me, thank you @andi34

andi34, 2019-09-19:
Thanks for the feedback @vdubuk .
@andreknieriem it's fine to submit the code change or it's a wanted behavior to stop the preview at the point it is?

andreknieriem, 2019-09-19:
Can be changed ;)

vdubuk, 2019-09-19:
Kiosk Mode  feature request

When running app in Kiosk mode it can be difficult to get to the Admin/settings page.  

therefore I would like to suggest of having a hidden shortcut to get to settings

index.php  around line 274 after the print mesg div

````
        
	<!-- added div --> 

	<div id="adminsettings" >
		<div style="position:absolute; bottom:0; right:0;"><img src="/resources/img/spacer.png" 
        alt="adminsettings" ondblclick="adminsettings()"></div>
	</div>  
 
	<script type="text/javascript" src="/resources/js/adminshortcut.js"></script> 

	<!-- end -->

	
     
```
create new file aresources/js/adminshortcut.js 

```
var admincount = 0;
		function countreset() {
			admincount = 0;
		}
		function adminsettings() {
			if (admincount == 3) {
				window.location.href="admin/index.php";
			}
			console.log(admincount);
			admincount++;
			setTimeout(countreset, 10000);
		}

```

add spacer.png to resources/img/ 
![spacer](https://user-images.githubusercontent.com/4080228/65237329-5ecd2200-dad2-11e9-9b86-efa306e6558d.png)

andi34, 2019-09-19:
> Kiosk Mode feature request
> 
> When running app in Kiosk mode it can be difficult to get to the Admin/settings page.
> 
> therefore I would like to suggest of having a hidden shortcut to get to settings
> 
> index.php around line 274 after the print mesg div
> 
> ```
>         
> 	 
> 
> 	<div id="adminsettings" >
> 		<div style="position:absolute; bottom:0; right:0;"><img src="/resources/img/spacer.png" 
>         alt="adminsettings" ondblclick="adminsettings()"></div>
> 	</div>  
>  
> 	<script type="text/javascript" src="/resources/js/adminshortcut.js"></script> 
> 
> 	
> 
> 	
>      
> ```
> create new file aresources/js/adminshortcut.js 
> 
> ```
> var admincount = 0;
> 		function countreset() {
> 			admincount = 0;
> 		}
> 		function adminsettings() {
> 			if (admincount == 3) {
> 				window.location.href="admin/index.php";
> 			}
> 			console.log(admincount);
> 			admincount++;
> 			setTimeout(countreset, 10000);
> 		}
> 
> ```
> 
> add spacer.png to resources/img/ 
> 
> [image]
> ```

works fine. ready for testing in my dev branch

andi34, 2019-09-19:
Double tap in the right bottom corner until the admin page appears (for me it takes 4 double clicks in chrome) 

[image]

andi34, 2019-09-20:
https://github.com/andi34/photobooth/commit/6b926f8d9118147e2f5679d647fac88fcdc30c28

Jemand eine Idee?

andi34, 2019-09-20:
Found the issue. config.json is ignored... Config.inc.php/my.config.inc.php is used only.
Need to find out how to solve it.
Maybe someone can help?

andi34, 2019-09-20:
Found a solution. Moved the php files out of the chroma_keying folder:
https://github.com/andreknieriem/photobooth/commit/f607a4196dbe60def3a0a6f3af0002ec9da1324f

[5 more replies]

closed 2019-10-11 by andi34