andreknieriem/photobooth #69: Create a FAQ

andi34, 2019-09-10:
## Collection of questions to create a proper FAQ
You noticed there's the same question again in the comments? Please let us know in the comments below if we haven't noticed it already.

### Is my Camera supported?
Some DSLR and Compact Cameras are not supported by this project. Please check for your specific model [here](http://gphoto.org/proj/libgphoto2/support.php).

sualko, 2019-10-02:
Can you move this to the wiki, so that everyone can change, update or add stuff?

andi34, 2019-10-02:
Sure, that was the goal. this issue was only used as a starting point to collect stuff first.

andi34, 2019-10-02:
Everything moved to Wiki.
Everyone can feel free to update pages or add more questions and answers.

Next step will be to move the installation instructions out of the README and add pages for every supported platform (Win,Linux,Mac).

closed 2019-10-02 by andi34

andi34, 2019-10-02:
Installation instructions from README moved to Wiki

sualko, 2019-10-06:
Why did you create a new page for every question? In my opinion it's clearer and more common to have every faq on one page. If you don't mind, I would delete all those pages and put them on one.

andi34, 2019-10-06:
Feel free to optimize it. Thought it's cleaner this way.