andreknieriem/photobooth #70: green screen keying auf raspberry pi

koenigd79, 2019-09-11:
Hallo,
erstmal toll wie ihr den Funktionsumfang der Box immer erweitert.

Eine Frage zum green screen keying:
Die Photobooth läuft bei mir inzwischen auf einem Raspberry Pi 4 mit 4gb mit angeschlossenem 13.3" Touchdisplay.
Das green screen keying kann nur aus der Gallery gestartet werden, soweit verstanden. Jetzt ist es so, dass meist das Bild anzeigt wird (machmal bleibt es schwarz bze. Fehlermeldung keying nicht möglich kommt) und die grünen Flächen sind schwarz, wenn ich jetzt auf einen der Hintergründe tippe zeigt es aber nur noch den Hintergrund an und nicht mehr das ursprüngliche Bild... wenn ich das ganze vom Handy/Tablett aus startet tut es.
Einer eine Idee woran das liegen könnte?? von der Rechenleistung her wird der Pi4 wenig gefordert.

Gruß Daniel

andi34, 2019-09-11:
Mal geschaut ob es am localhost vs IP addressen Aufruf liegt?
Keying geht nur dann, wenn er entsprechend eine verkleinerte Kopie des Bildes im keying Ordner findet. Auflösung musste reduziert werden damit das ganze funktioniert.

koenigd79, 2019-09-11:
Hi, jetzt einiges probiert. Egal ob localhost, IP oder Hostname im chromium wird immer der Hintergrund "davor" gelegt... Bei Zugriff per Rechner, Handy Tablet funzt es...

andi34, 2019-09-12:
Ich schaue mir das ganze nochmal an am Pi die Tage.
Am großen PC funktioniert es direkt ohne Probleme.
Ggf. Müssen wir nochmal die Auflösung runterschrauben, kommt der Pi vllt nicht zurecht mit im Browser.
Andere Idee habe ich zur Zeit nicht wirklich spontan.
Höchstens noch https://github.com/andreknieriem/photobooth/blob/master/takePic.php#L271

Mach Mal aus
```
	imagejpeg($source, $filename_keying, 100);
```
Bitte ein
```
	imagejpeg($source, $filename_keying, 80);
```
Und versuchs nochmal mit einem neu geschossenem Bild anschließend. Das schraubt die jpeg Qualität etwas runter.

andi34, 2019-09-12:
Nebenfrage: geht es über einen anderen Browser?

koenigd79, 2019-09-12:
Codeänderung zeigt keinen Erfolg. Das Bild wird aufbereitet, die vormals grünen Flächen sind schwarz, also wahrscheinlich Durchsichtig.  Bei Klick auf eines der Hintergrundbilder wird das Hibtergrundbild aber vollständig vor das gemachte Foto gelegt.
Habe GPU Speicher auch mal auf 512 MB geändert, zeigt aber keine Änderung.
Andere Browser, es ist noch epiphany installiert, da läuft irgend ein Prozess amok, erst mit Beenden des Browsers geht die Prozessorauslastung zurück.
Soweit die kurzen Tests vor der Arbeit.
Gruß Daniel

andi34, 2019-09-12:
Ich schau dann selbst Mal in Ruhe.
Bin relativ platt und wenig Zeit wegen Umzug meiner Freundin mit Tochter zu mir. Noch viel zu organisieren etc 🙈

Ich sag bescheid sobald ich was weiß.
Vllt irgendwelche Einstellungen im Chromium die default nicht gut sind.

andi34, 2019-09-20:
Hier das selbe am Pi. Suche noch nach einer Loesung.

andi34, 2019-09-27:
Leider bisher keine Loesung gefunden. Ziemlich komisch.
Jemand eine Idee?

msmedien, 2019-09-30:
Hallo, mir ist bei dem Green-Screen-Feature etwas aufgefallen... Wenn ich ein Foto mit grünen Hintergrund/Elementen erstelle, wird es ganz ordentlich vom ausgeschnitten und schwarz markiert - soweit so gut. Drücke ich nun auf einen der möglichen Hintergründe, wird das eigentliche Foto vollständig überlagert, anstatt in die nun transparente Fläche eingesetzt zu werden. - Soweit waren wir sicherlich alle schon einmal. Drücke ich aber 50 mal nacheinander auf der gleiche Hintergrundbild, wird es tatsächlich eingearbeitet wie gewünscht. (Geht auch wenn man 50 mal durch die Hintergründe hin und her wechselt) Drücke ich dann zum 51. mal wird das Foto wieder komplett überlagert. Jetzt muss man wieder exakt 50 mal klicken bis es problemlos funktioniert. Es sind immer exakt 50 klicks nötig.

Es ist reproduzierbar und somit eigentlich lösbar, oder nicht? Ich bin kein Programmierer aber vielleicht wurde statt eines Timers ausversehen eine Schleife eingesetzt?

Das System läuft auf einem Pi4 mit 4GB Ram und den letzten Updates.

koenigd79, 2019-10-02:
Hi,
habe gerade meine installation auf den aktuellen Stand gebracht und siehe da es funktioniert. Pi4 mit 4 GB, akutelles System und aktuelles Photobooth. Bild wird entsprechend ausgeschnitten und beim click auf einen der Hintergründe wird dieser sofort eingebelendt, direkt am Montior am RPI. Also aus meiner sicht scheint es zu funktionieren, wer kann das bestätigen??
Gruß Daniel

msmedien, 2019-10-04:
Also ich habe meine Installation auch gerade geupdated, allerdings bleibt bei mir das Problem mit den 50 Klicks bestehen...

Für das Update habe ich folgende Befehle genutzt und dann Cache vom Chromium geleert:
sudo apt-get update
sudo apt-get dist-upgrade

Oder bedarf es da noch weiterer Befehle? Bin bzgl. Raspbian eher Laie. ^^" Wenn es jetzt bei Dir problemlos geht, dann sollte es doch bei mir auch funktionieren können...?

msmedien, 2019-10-04:
@andi34 Ja, ich habe gerade nochmal die aktuellste Version der Photobooth aufgespielt. Das Ergebnis ist aber das Gleiche... :-/

andi34, 2019-10-04:
Sowohl beim Zugriff von externen Geräten (z.B. Handy) als auch am Pi selbst?
Ich habe das Problem nur am Pi gehabt bisher, bei Aufruf vom Handy ging zuletzt alles problemlos.

msmedien, 2019-10-04:
@andi34 Vom Handy aus funktioniert es tadellos, aber das wurde ja schön öfter festgestellt. :)

Wichtiger für die Gäste wäre es natürlich, wenn es sofort vom PI aus klappt. :P Das mit den Berechtigungen muss ich später einmal ausprobieren, da ich gleich noch ein paar Termine habe. - Aber wenn es nur an den Berechtigungen liegen würde, dürfte es vom Handy doch auch nicht gehen, oder?

andi34, 2019-10-04:
Richtig, aber man hat schon Pferde vor der Apotheke kotzen gesehen 🙈 von daher wird es nicht schaden und wenn der Rest weiterhin funktioniert ist das Setup im ganzen deutlich sicherer.

msmedien, 2019-10-04:
@andi34 @koenigd79 Ich habe das Problem jetzt gefunden! :) Zumindest funktioniert es bei mir seither wunderbar.

Einfach bei Chromium das Flags-Menü aufrufen. (In die Adresszeile chrome://flags eintragen.) Dort dann den Punkt "Accelerated 2D canvas" auf "Disabled" stehen. Browser neustarten und zack läuft das Ganze perfekt. :)

Ich hoffe das ich damit kein Einzelfall bin und es erstmal ein langfristiger Lösungsansatz ist. :)

andi34, 2019-10-04:
Super, vielen Dank @msmedien ! Werde ich testen sobald als möglich

andi34, 2019-10-04:
Added to Wiki. https://github.com/andreknieriem/photobooth/wiki/Chromakeying
If someone else confirms it's working we can close the issue.

koenigd79, 2019-10-09:
Hallo,
habe jetzt in den Einstellungen geschaut und der Punkt "Accelerated 2D canvas" auf "Enabled" und es fnktioniert trotzdem. Werde es dann jetzt tortzdem auf disabled setzten.

Gruß Daniel

closed 2019-10-09 by andi34