andreknieriem/photobooth #71: Undefined variable & Invalid arguement

vdubuk, 2019-09-12:
Unfortunately I am still experiencing the following errors and unable to resolve. I would appreciate any assistance in resolving.

PHP Notice:  Undefined variable: date in /var/www/html/index.php on line 166, referer: http://127.0.0.1/

PHP Notice:  Undefined variable: config in /var/www/html/folders.php on line 4, referer: http://127.0.0.1/

PHP Warning:  Invalid argument supplied for foreach() in /var/www/html/folders.php on line 4, referer: http://127.0.0.1/

andi34, 2019-09-13:
https://github.com/andi34/photobooth/commit/9b49e07090997946482e8aa8badc77f2f5ae04a5

Should address one if the issues

andi34, 2019-09-13:
Both Notice & Warning should be gone now.

closed 2019-09-13 by andi34