andreknieriem/photobooth #72: Bug: Collage Sometimes not created

andi34, 2019-09-19:
Mentioned in Andrés Webpage.

Idea to solve this problem:
Adjust the busy time, make if configurable:

https://github.com/andreknieriem/photobooth/blob/master/resources/js/core.js#L137

Trying a higher value might solve the problem.

flighter18, 2019-09-20:
While I was implementing the collage function I had some trubles with PHP max_execution_time?
I didn't have any problem last weekend during the wedding.

andi34, 2019-09-20:
@flighter18 like this as ```/var/www/html/php.ini``` ?

```
[PHP]
max_execution_time = 60
max_input_time = 60
memory_limit = 128M
```

andi34, 2019-09-20:
forget it, found it: https://github.com/andreknieriem/photobooth/commit/8fa282d3cf93d0e6b56e5c77b50f5452437744dd

open http://localhost/phpinfo.php

take a look for "Loaded Configuration File", you need  sudo rights to edit the file.
Page will look like this:

[image]

closed 2019-09-27 by andi34