andreknieriem/photobooth #73: Photobooth only works for the first picture.

Daveisbrave, 2019-09-20:
Die Fotobox macht nach dem Start ein Foto (Einzelfoto-Modus) und speichert es korrekt ab. 
Klickt man ein weiteres Mal auf den Foto-Button, kommt der Timer, Cheese und dann die Fehlermeldung. (Versuche es ein noch einmal.) In der Konsole wird kein Error angezeigt.

andi34, 2019-09-20:
Ein bisschen mehr Infos?

Welches Geraet?
Welches Betriebssystem?
Kamera-Akku voll?
Wird an der Kamera ein Problem angezeigt?

andi34, 2019-09-20:
Achso und wichtig: Kamera darf im zweifel auch nicht in den Standby gehen.

andi34, 2019-09-20:
Can't reproduce. Not on Xubuntu Desktop PC and not on Pi4 running lastest Rapsbian.

andi34, 2019-09-20:
Bist du David aus den Kommentaren hier https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html ?

Falls ja: Text on Print an oder aus?
Falls an: ausschalten behebt das Problem?

Im Zweifelsfall https://github.com/andreknieriem/photobooth/blob/master/resources/fonts/GreatVibes-Regular.ttf anklicken und auf "Installieren" klicken.

Ggf. Musst du noch ein paar Codezeilen anpassen:

https://github.com/andreknieriem/photobooth/blob/b05af90b5fa89707d900da6582c09cc97594175b/print.php#L80-L82

```
'resources/fonts/GreatVibes-Regular.ttf'
```
Dann ggf. ersetzen mit
```
'GreatVibes-Regular.ttf'
```

Oder einer anderen beliebigen Windows Font die bereits vorhanden ist.

Daveisbrave, 2019-09-20:
Danke für die schnelle Hilfe. Ich bin tatsächlich dieser David. 
Die angefragten Infos:
Welches Geraet? Nikon D7000 per USB an einem Windows Tablet (Windows 10, 32 Bit mit XAMPP und digicamcontrol in 32 Bit Version. --> Xampp gibt es in der neuesten Version nicht mehr in 32 Bit).
Kamera-Akku voll? Ja, SD Karte leer.
Wird an der Kamera ein Problem angezeigt? Die Kamera zeigt an "PC", also extern gesteuert. Nach dem ersten Bild ist sie "besetzt", kann nicht mehr auslösen, auch manuell. Nach Neustart des Systems inkl. Kamera kann wieder ein Bild gemacht werden. 

Der Font-Fehler wird nicht mehr angezeigt. Dafür: "Das Bild an URL: http://localhost/images/[...].jpg konnte nicht mit MIME Image/tiff kodiert werden. 

und ein weiterer Error:
JSON.parse Error: Invalid Character at Position:1 core.js (145,12)

Daveisbrave, 2019-09-20:
Habe möglicherweise das Problem gefunden: Die Kamera hat zwei SD Slots und speichert normalerweise auf einer die RAW-Datei, auf der anderen die jpg. Habe jetzt auf "jpg-only" gestellt und die Größe verringert. Sieht vielversprechend aus.

andi34, 2019-09-20:
Danke für die Rückmeldung :)
Sag bitte nochmal bescheid ob gelöst oder nicht gelöst ;)

Daveisbrave, 2019-09-20:
Gelöst!

closed 2019-09-20 by Daveisbrave

andi34, 2019-09-20:
Danke 🙂

andi34, 2019-09-20:
Achso: irgendwas noch nicht dokumentiert im README was Windows spezifisch ein/umgestellt werden muss?