andreknieriem/photobooth #81: Project structure

sualko, 2019-09-26:
From my point of view the current project structure could be enhanced to have a better separation between front- and backend. I suggest the following: We put all frontend related stuff into `resources/`. Therefore we move the `lib/` folder to the root folder. All those folders with user generated content (`images/`, `keying/`, ...) should go into it's own folder. To be able to protect the config files with a htaccess file, it would be good to move also those two files to its own dir. As a result we would have something like this:

- admin/
- chroma_keying/
- resources/
  - lang/
  - css/
  - fonts/
  - img/
  - js/
  - sass/
  - lang/
- data/
  - images/
  - keying/
  - print/
  - qrcodes/
  - thumbs/
  - tmp/
  - data.txt
- vendor/
  - PHPMailer/
  - phpqrcode/
- config/
  - config.inc.php
  - my.config.inc.php
- gulpfile.js

What are your two cents? Does such a change make sense?

andi34, 2019-09-26:
@sualko I recently had to move the chromakeying php files down to root as it was ignoring the admin/config.json

If you have a better solution I'd be on your side regarding the structure.

sualko, 2019-09-26:
As I didn't move the chroma_keying dir, there should be no issue. But in general this change is not urgent.

andi34, 2019-09-26:
👍 from my side

andi34, 2019-09-28:
https://github.com/audstanley/NodeJs-Raspberry-Pi
Keeping this here in case it's interesting.

Not sure why, but on a different booth project node v9 is needed ( https://github.com/philipptrenz/photo-booth/commit/a3b2314632a9174f7d69abed2327b2cb34d87047#diff-d6c5855a62cf32a4dadbc2831f0f295f )

mhellmeier, 2019-10-19:
Just a quick question on this closed issue:
Is there any reason for the _vendor_ directory in the version control system?
If you are looking into the recommandations from composer, the best practice is not committing the directory to git (https://getcomposer.org/doc/faqs/should-i-commit-the-dependencies-in-my-vendor-directory.md).

andi34, 2019-10-19:
PHPmailer is a submudule and only a link here.

phpqrcode is not on GitHub if I am not wrong and can't be added as submodule.
 Afaik @sualko wanted to rework the QR code generation in future too anyway.

andi34, 2019-10-19:
Well, @sualko we could also add this as submudule maybe https://github.com/t0k4rt/phpqrcode

sualko, 2019-10-21:
> Is there any reason for the vendor directory in the version control system?

That's right, but in this special case it's no big deal. As @andi34 said, it's source is on sourceforge and we want to change the qr part anyway.

> we could also add this as submudule maybe https://github.com/t0k4rt/phpqrcode

As far as I can tell, this is only a mirror of http://phpqrcode.sourceforge.net, but I don't want to distribute code from unknown sources. Sure we could compare if that's the same code, but than we have to do that on every change. At least this is nothing I would like to do :smile:.

PR #90 (2019-10-10): Enhance project structure
for more details, see #81

- [x] reduce number of global vars for config
- [x] merge master

close #81, #82 

Also includes new features:
- gallery standalone (`gallery.php`)
- add button to delete picture after picture was taken and displayed on the screen
- change style via admin panel
- add trigger keys via config
- add option to force the use of a buzzer
- add option to enable CUPS button
- add option to resize and crop image by center at print
- use same printer settings for chromakeying as for normal prints
- verify that config and data folder is writable
- take [... cut]