andreknieriem/photobooth #82: Resizing image for printing

plin2, 2019-09-26:
I'm using a Canon SELPHY CP910 with a paper format of 100x148 mm². The pictures taken are usually in 4:3 Format. The print option "fit-to-page" causes the printout to be distorted. 

To fix this problem I installed imagemagick and created a cropping script.

In config.inc.php I adjusted the print command
$config['print']['cmd'] = '/var/www/html/prep_and_print.sh %s';

/var/www/html/prep_and_print.sh:
#!/bin/bash

# adjust these parameters for your printer
printer_name="Canon_CP910"
printer_max_width=1868
printer_max_height=1261

# fetch parameters
foto=$1

rm images/print.jpg
mydate=`date`

# calculate cropping 
wid_hig=`identify $foto`
wk1="${wid_hig##*JPEG }"
wk2="${wk1%% *}"
is_wid="${wk2%%x*}"
is_hig="${wk2##*x}"

let is_ratio=1000*$is_wid/$is_hig
let tgt_ratio=1000*$printer_max_width/$printer_max_height

echo $is_ratio $tgt_ratio

if [ $is_ratio -lt $tgt_ratio ]
then
   # crop height
   tgt_width=$is_wid
   let tgt_height=1000*$is_wid/$tgt_ratio
   let y_crop=($is_hig-$tgt_height)/2
   x_crop=0
else
   # crop width
   let tgt_width=$is_hig*$tgt_ratio/1000
   tgt_height=$is_hig
   y_crop=0
   let x_crop=($is_wid-$tgt_width)/2
fi

convert -crop ${tgt_width}x${tgt_height}+$x_crop+$y_crop $foto images/print.jpg

sudo lp -d $printer_name -o landscape -o fit-to-page images/print.jpg

andi34, 2019-09-27:
Thanks for your feedback :) you've only problems if QR Code is disabled for print? Or in any conditions?

andi34, 2019-09-27:
Moved to https://github.com/andreknieriem/photobooth/issues/68 ,
Ready for review.

To test this changes (and everything else currently open for review)
Make sure your local clone does ignore filemode changes:
```
cd /var/www/html
git config core.fileMode false
sudo git config core.fileMode false
```
Now add my Github copy as remote and fetch the source:
```
git remote add andi34 https://github.com/andi34/photobooth
git fetch andi34
```
Make sure you have no local changes which will be lost, if you have you need to commit them. Best ist to keep them in a seperate branch you can go back to later:
```
git add --all
git commit -a -m "All my local changes"
git checkout -b myBackupBranch
```
Now switch to my source:
```
git checkout andi34/dev
```

You might need to set the permissions again on ```/var/www``` as mentioned in the README.
Important: Make sure deleting ```admin/config.json``` if exist!

In case you need to switch back to your local source you only need to checkout to your backup branch which we've created before.
```
git checkout myBackupBranch
```

Questions? Let me know!

plin2, 2019-09-27:
> Thanks for your feedback :) you've only problems if QR Code is disabled for print? Or in any conditions?

yes, even without QR code I experience a distortion. The printout is stretched horizontally.

The approach in my script:
- anything higher than 2:3 will be cropped on top and bottom
- anything broader that 2:3 will be cropped on left and right

andi34, 2019-09-27:
You might want to play around with the option I've added. You only need to follow mentioned steps. Guess it'll solve your problem.

plin2, 2019-09-28:
> 
> 
> You might want to play around with the option I've added. You only need to follow mentioned steps. Guess it'll solve your problem.

Well, not really. Now I get a picture in the size given in config.inc.php. Since the orginal has a 4:3 aspect ratio and the target is 1000x500, the picture (aspect ratio 2:1) is filled with a white box on the right hand side.  My method is not going for a target size but rather taking the maximum from the original and just cutting off some excess content, based on the aspect ratio of the printers paper.

Examples

original size: 2592x1944
aspect ratio 4:3
cropping on top and bottom: 47 px
cropped size: 2592x1850
aspect ratio 3:2

original size: 1920x1080
aspect ratio 16:9
cropping on left and right: 150 px
cropped size: 1620x1080
aspect ratio 3:2

andi34, 2019-09-28:
You can change the value for your needs. Made it optional. It'll crop from center based on given values. 
Make changes through config or via admin panel.

```
$config['crop_width'] = '1000';
$config['crop_hight'] = '500';
```

plin2, 2019-09-29:
ok, I can live with that. I would set it according to the image size of the attached camera and the cropping I want to take place.
If you add the feature to switch between cameras it would be nice to switch this setting along with the camera :-).

andi34, 2019-10-02:
Reopen until changes are merged.

sualko, 2019-10-10:
Feature merged now.

closed 2019-10-10 by sualko