andreknieriem/photobooth #84: responsive theme : Auto fonts size

thymon13, 2019-09-28:
Hi,

Is it possible to adjust the Fonts size ?

[image]

andi34, 2019-09-28:
What's your resolution?

thymon13, 2019-09-28:
Responsive does not adapt the font size automatically?
1280x800

andi34, 2019-09-29:
For now you could adjust the font-size inside the style.css (https://github.com/andreknieriem/photobooth/blob/master/resources/css/style.css#L60)
```
h1 {
  font-size: 50px;
  margin: 33px 0; }
```
Try something below 50px.

andi34, 2019-09-29:
@sualko could you take a look at it?

thymon13, 2019-09-30:
same in MacBook with Google Chrome
[image]

sualko, 2019-09-30:
Please download my fix from https://github.com/sualko/photobooth/tree/fix/start-layout and report if it fixes your issue.

PR #89 (2019-10-06): fix start screen layout
fix #84
commit 4dcbe2ec (2019-10-06): fix start screen layout

fix #84
closed 2019-10-06 by sualko