andreknieriem/photobooth #87: 1.9.0  preview with device cam - not working/displaying

vdubuk, 2019-09-29:
Testing version 1.9.0 this afternoon, I found that the preview with device cam is not working.

andi34, 2019-09-29:
Anything in the console?
Press F12 to see the console.

obkram, 2019-09-29:
Danke erstmal für dieses Tolle Release.
Ihr macht einen super Job.
Bei mir läuft das Device Cam Preview ebenfalls nicht. Hab leider schon alles wieder abgebaut. F12 werde ich testen.
Bei den Farbfiltern bleiben bei mir alle Filter mit bluer im Namen ebenfalls ohne Effekt.

andi34, 2019-09-30:
@vdubuk and @obkram please test https://github.com/sualko/photobooth/commit/2b2842663b11428bd0177fca730bbf09d3f8ad7e

obkram, 2019-09-30:
Ich habe es nun getestet. Zumindest bei mir funktioniert es nicht. Bei der Consolenausgabe via F 12 Taste bekomme ich folgende Ausgabe:
Could not get User media: DOMException core.js:127

Ich hoffe das hilft euch irgendwie weiter.

andi34, 2019-09-30:
@obkram bitte die Lösung von oben versuchen und mitteilen ob es damit geht

obkram, 2019-09-30:
Die Lösung von oben hatte ich bereits vor meinem Test integriert.
Die Meldung aus der Console in Zeile 127 kommt ja aus dem neuen File.

sualko, 2019-09-30:
@obkram can you post a screenshot from the error message? Can you unfold the error message? Because it's a bit short. Which browser and version do you use?

sualko, 2019-09-30:
Do you access the site via http and a ip? Chrome requires to use https if you like to get the webcam stream. There is only an exception for localhost. Just guessing.

obkram, 2019-10-01:
Das hat mir keine Ruhe gelassen. Ich habe heute morgen nochmal rumprobiert. Jetzt gehts. Zumindest in meinem Fall muss ich die USB Kamera, sofern sie vor dem Booten des RPI am USB Port eingesteckt war, aus und wieder einstecken. Danach geht es. Nach dem nächsten Neustart das gleiche. Es funktioniert dann sowohl mit der alten als auch mit der neuen Konfig.
@ Sualko: You are right, the preview only works with localhost or 127.0.0.1

andi34, 2019-10-01:
Also Funktioniert es auch ohne die Codeänderung? Aber eben nur über localhost/127.0.0.1?
Ggf kompillier dir usbreset (Antwort von Li Lo hier https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line ) oder nutz das Python Script von hier https://github.com/mcarans/resetusb/ , ggf. damit kannst du das un- und replug vermutlich sparen.

sualko, 2019-10-01:
> Also Funktioniert es auch ohne die Codeänderung?

My fix is required.

obkram, 2019-10-01:
@sualko You are right
@andi34 : ich komme mir fast vor wie ein DAU, also das System zurück auf 0 und nochmal neu. Hatte vorher im Hintergrung noch Motioeye laufen um damit den Livestream für das Vorschaubild auf dem Tablet zu erzeugen. Von daher habe ich nie wirklich erkennen könne was die LED an der Kamera macht. Also nun ist sie standartmäßig aus, da keine Instanz darauf zugreift. Sualko hat recht, das Script muss auf jeden Fall geändert werden.
Ich arbeite mit dem Standartbrowser Chromium um die Funktion zu testen. Ledigtlich bei 127.0.0.1 wird die Kamera aktiviert und dann kommt auch die Abfrage ob Chromium auf die Kamera zugreifen darf. Chromium trägt die Kamera dann in die whitelist ein.
Um Missverständnissen vorzubeugen. Die Funktion habe ich derzeit auf dem Pi getestet. Ich werde sie aber später auf dem Tablet ( IPad ) nutzen müssen. Da funktioniert es allerdings nicht, aber das ist hier gerade auch nicht Thema. Vielen Dank schonmal an euch Beide.
@vdubuk : is the preview now running on your pi too?

obkram, 2019-10-01:
Das Problem mit dem Neustecken der USB Kamera lag am Motioneye. Das funktioniert jetzt auch super.

obkram, 2019-10-01:
Eine bescheidene Frage hab ich noch, dann gebe ich auch Ruhe. So wie auf dem Bild zu sehen habe ich es derzeit auf meiner Box das Vorschaubild mit der USB Kamera am Pi via Motioneye als Hintergrund laufen. ( siehe Pull #20 )
Es macht den Gästen am meisten Spaß sich anhand des Livebildes die Masken, Photoprops etc. auszusuchen. Hat auch den Vorteil das man immer sieht ob man im Bild steht oder nicht. Wäre das nicht auch für euch so machbar? ich bekomme es in die neue Architektur mit meinem gefährlichen Halbwissen nicht mehr implementiert. ( Den Emoji natürlich nicht )
[image]
 Gruß Sven

obkram, 2019-10-02:
@sualko : since all our changes from yesterday i found a new issue. I checked it on a blank system too and there the issue doesn´t exist.
When I take a picture and no camera is connected...
1. Chee....se
2. Error message : Es ist ein Fehler aufgetreten....
3. The system says that the picture is in progress. From there is no way back.

Maybe the issue is only on my system but if it is possible to you please check this on your system befor a new version of Photobooth will be uploaded. Thank you.

andi34, 2019-10-02:
@obkram you're adding the changes by hand?
Easier to cherry-pick the changes on top of current master branch from here

```
git fetch https://github.com/sualko/photobooth fix/webcam
git cherry-pick 2b2842663b11428bd0177fca730bbf09d3f8ad7e
```

sualko, 2019-10-02:
> Was auch ziemlich cool wäre wäre wenn man auf der Startseite auf "Foto erstellen" klickt und dann auf sie Seite mit dem Liveview kommt.

Could you open a new issue? Otherwise this cool feature could be forgotten.

> Maybe the issue is only on my system but if it is possible to you please check this on your system

Please open a new issue and add a description on how to reproduce your issue.

obkram, 2019-10-02:
@andi34 : of course, adding the changes by hand. I don't know this witch fetch and cherry-pick.
The console doesn't show any errors.

vdubuk, 2019-10-02:
> @vdubuk and @obkram please test [sualko@2b28426](https://github.com/sualko/photobooth/commit/2b2842663b11428bd0177fca730bbf09d3f8ad7e)

The suggest code edits work for me.  thank you

andi34, 2019-10-02:
@obkram sure the issue is related to this? If not I guess the change should me merged.

andi34, 2019-10-03:
https://github.com/andreknieriem/photobooth/blob/6a066778f3f4deb5bb64826cd6088ac50e42ab2c/resources/js/core.js#L193-L199

If there's an error you should get a error message and a Button to reload the page.
So the issue is independent. Going to merge this change here.
@obkram if the issue still exist after merge please open a new issue.

PR #88 (2019-10-03): fix webcam preview
fix #87
commit 2b284266 (2019-10-03): fix webcam preview

fix #87
closed 2019-10-03 by andi34