andreknieriem/photobooth #91: Probleme mit Chroma Keying nach Fotoerstellung

obkram, 2019-10-01:
Mir ist folgendes aufgefallen. Hat man ein Foto aufgenommen dann bekommt man diese nach der Aufnahme angezeigt. Wenn man aus dieser Ebene auf den Button Gallerie klickt, das gerade aufgenommene Bild wählt und dann von dort aus Chroma Keying ( also das Pinselsymbol ) auswählt dann bekommt man die Meldung...
Chroma Keying nicht möglich!
Darunter befindet sich ein Schließen Butten. Dieser ist allerdings 15-20 Sekunden inaktiv die Meldung bleibt so lange stehen bis man dann nach 20 Sekunden den Button drückt.

Geht man allerdings unmittelbar nach der Bildaufnahme via Home Button zurück ins Hauptmenü und wählt von dort aus die Galerie, das letzte Foto und dann Chroma Keying dann funktioniert es direkt.

andi34, 2019-10-01:
@obkram ist mir auch aufgefallen, hatte noch keine Zeit mir das anzuschauen.

obkram, 2019-10-01:
@sualko : in my case it works fine, thank you
@andi34 : in diesem Zusammenhang ist mir aufgefallen das das erste Bild ins leere läuft. Soll heißen es wird nicht als Bild gewertet und kann auch nicht selektiert werden. Das funktioniert dann erst wenn zwei Bilder in der Gallerie sind.
[image]

sualko, 2019-10-01:
@obkram I fixed your second issue also the pr. Please give it another try.

obkram, 2019-10-01:
@sualko : I tested it, works fine too.

obkram, 2019-10-01:
@sualko : i know that this is the wrong ticket but we have a nice run ;-).
Do the picturefilter **blur** and **selective blur** have any effect on your system?

andi34, 2019-10-01:
@obkram seems like the blur of GD isn't that effective.
https://stackoverflow.com/questions/7245710/php-gd-better-gaussian-blur
https://stackoverflow.com/questions/14428257/how-to-achieve-a-blur-effect-in-php

We would need to use a different setup for the blur to be not that weak.

commit a288562f (2019-10-01): fix image link after capture

fix #91
PR #94 (2019-10-01): fix image link after capture
I think the new parameter is not used anywhere, so I simply removed it.

@obkram can you test this fix?

fix #91
closed 2019-10-01 by andi34