andreknieriem/photobooth #92: Capture raw images

sualko, 2019-10-01:
If the camera is set to record raw or raw+jpg images, gphoto downloads only the raw file with jpg as file extension which results in an faulty image and therefore some errors. I guess adding `--keep-raw` as option to gphoto should fix that. Will open a pull request as soon as I gave it a try. Another option would be to use `%C` as file extension, so that gphoto downloads both versions with the correct extension.

thymon13, 2019-10-01:
I use `--keep-raw --capturetarget=1`

Raspberry captures the JPEG image and deleted it from the camera. But leave the RAW files for later processing

andi34, 2019-10-02:
Gphoto2 supports many cameras and not every camera has RAW. We might want to update the FAQ and tell possible ways to fix issues.
Or: add an option to choose different config for different camera models? This can be expanded.

sualko, 2019-10-02:
gphoto ignores the option if the camera doesn't support raw. Therefore I think it's save to add this to the default command.

andi34, 2019-10-02:
Haven't tried if it hurts. Maybe someone with a supported can without raw can confirm it working

sualko, 2019-10-21:
I added the information to the wiki and with the improved error handling, I think it's ok now. Therefore I close this issue.

closed 2019-10-21 by sualko