andreknieriem/photobooth #93: Add description to auto-unmount camera

sualko, 2019-10-01:
The current documentation recommends to delete the gvfs services which is a pretty rough action. I think it would be better to automatically unmount the camera, like described in https://askubuntu.com/a/301133. If #86 is merged, I will create a pr.

andi34, 2019-10-05:
Another solution found here instead deleting the services https://askubuntu.com/questions/673259/restricting-mtp-devices-from-being-mounted-by-a-user-group

### Blocking MTP and PTP
Enter this location as root ```/usr/lib/gvfs```

"sudo nautilus" will be easy if you tend to use GUI
```sudo nautilus``` and then navigate to ```computer/usr/lib/gvfs```

There will be list of services Remove the execute permission by right clicking the appropriate service-->properties-->permission-->unckeck "execute" and change the read only access from "read only" to "none".

Remove the execute permission for
```
gvfsd-mtp
gvfs-mtp-volume-monitor(for safer side)
gvfsd-gphoto2
gvfs-gphoto2-volume-monitor(for safer side)
```

andi34, 2019-10-07:
`sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor`

Is enough. Only needs a reboot after removed execution permissions.

andi34, 2019-10-08:
Updated the wiki already: https://github.com/andreknieriem/photobooth/wiki/_compare/f01d65ac38fa4128f169fc5316a47064de5e19c6...569d85519aabdbb1f58f5f2cd19753449244d913

closed 2019-10-09 by sualko