andreknieriem/photobooth #95: reset mailaddresses.txt via resetbutton

obkram, 2019-10-01:
Wäre es machbar das Löschen des Inhalts der mailaddresses.txt mit dem Zurücksetzen Button zu verbinden?

obkram, 2019-10-01:
sorry, that doesen´t work. I can see that the data.txt will be removed when i click on the reset button but the mail-addresses.txt will still exist.

sualko, 2019-10-01:
Do you have any errors in your apache log? Which permission has `mailaddresses.txt`? Easiest way is to execute `ls -l mailaddresses.txt ` in your root folder.

obkram, 2019-10-01:
@sualko : sorry, i copied the lines to the Wrong position. Now it works fine. Thanks.
Should i close the ticket?

sualko, 2019-10-01:
Please leave it open. If the pr is merged, it's closed automatically and marked as resolved by that pr.

PR #97 (2019-10-01): delete mail-addresses.txt on reset
fix #95
commit e85b0a62 (2019-10-01): delete mail-addresses.txt on reset

fix #95
closed 2019-10-01 by andi34