andreknieriem/photobooth #96: enhance image filters

sualko, 2019-10-01:
For me there are currently two issues regarding image filters:

1. Personally I like consistency across images, so an option would be nice to specify a filter for all images.
2. Filters are very basic and doesn't look that good.
3. If you have a selection dialog for users, there should be a preview.

I think the first point is easy to implement, but for the second we should probably switch from GD to ImageMagick. Would love to have a easy way to specify own filters in the config, but I'm not sure how easy this is. There is also a nice starting point: https://code.tutsplus.com/tutorials/create-instagram-filters-with-php--net-24504

@andi34 @obkram what do you think?

This would go on top of the current refactoring branch. @andi34 therefore please don't rebase this branch :wink:.

[image]

andi34, 2019-10-01:
I've played around with simple imagemagick filter too but noticed the performance wasn't good, that's why I used the GD way.
But I haven't tried PHP ImageMagick Extension (IMagick)

andi34, 2019-10-11:
Guess we can close https://github.com/andreknieriem/photobooth/issues/111 and keep this here depending on how the rework looks like later?

sualko, 2019-10-13:
> Guess we can close #111 and keep this here depending on how the rework looks like later?

I think it's a good idea to close that issue and include it in 2.1. I think it make sense to release the 2.0 as soon as possible.

andi34, 2019-10-19:
https://github.com/andreknieriem/photobooth/pull/149 related to **1.**

andi34, 2019-11-09:
https://phpimagick.com/Imagick

Leaving this here in case imagick gets interesting in future. The webpage is available on GitHub too, effect examples can be seen there.

rawbertp, 2019-11-10:
How much effort would it be to change the UI-flow as follows:

1. Take picture
2. Display the picture (file saved automatically)
3. Allow users to apply a filter
4. The pic changes according to filter selected
5. If the users want to save the image with the applied filter they would have to click a "Save" button (TBD: would this overwrite the original picture or will both be saved?)
6. For performance reasons it might be necessary to scale the image down for the filter preview and first apply it to the actual full-size image upon save was clicked

andi34, 2019-11-10:
@rawbertp depending on coding skills and time you can spend. For me this would be a big project without a known end result. Someone with advanced coding skills and knowledge might have it easier.

rawbertp, 2019-11-10:
@andi34 Appears to be much easier than expected...  This is the result with only a few lines of code changed. The missing 20% might be the hardest part, though...

[image]

andi34, 2019-11-10:
Wow, Looks great :)

andi34, 2019-11-11:
https://github.com/zaachi/PHP-Instagram-effects/blob/master/src/Image/Filter.php

In case there's anyone liking to take a look at