andreknieriem/photobooth #98: Live-Cam Wallpaper/ Countdown

obkram, 2019-10-02:
So wie auf dem Bild zu sehen habe ich es derzeit auf meiner Box das Vorschaubild mit der USB Kamera am Pi via Motioneye als Hintergrund laufen. ( siehe Pull #20 ) Das ganze geht dann via Wlan ins IPad.
Es macht den Gästen am meisten Spaß sich anhand des Livebildes die Masken, Photoprops etc. auszusuchen. Hat auch den Vorteil das man immer sieht ob man im Bild steht oder nicht. Wäre das nicht auch für euch so machbar? ich bekomme es in die neue Architektur mit meinem gefährlichen Halbwissen nicht mehr implementiert. ( Den Emoji natürlich nicht )
[image]

> **Zitat devnull1975:**
> Was auch ziemlich cool wäre wenn man auf der Startseite auf "Foto erstellen" klickt und 
> dann auf die Seite mit dem Liveview kommt. Dann könnte man den halbtranspareten Banner 
> evtl. ausblenden der bei dem Liveview etwas stört.

andi34, 2019-10-02:
https://github.com/andreknieriem/photobooth/issues/69

Steht schon fürs FAQ im Wiki drin

obkram, 2019-10-02:
> **Zitat sualko:**
>    Was auch ziemlich cool wäre wäre wenn man auf der Startseite auf "Foto erstellen" klickt und dann auf sie Seite mit dem Liveview kommt.
> **Could you open a new issue? Otherwise this cool feature could be forgotten.**

Sorry, vermutlich fehlinterpretiert. Soll ich den Fall schliessen?

andi34, 2019-10-02:
Lass ruhig offen, aber mit der Frage optional per Config aktivierbar zu implementieren.
Die Antwort war aber auf deine Frage ;)

> ich bekomme es in die neue Architektur mit meinem gefährlichen Halbwissen nicht mehr implementiert. ( Den Emoji natürlich nicht )
> [image]
>

andi34, 2019-10-11:
Reopen and close collection thread

obkram, 2019-10-15:
Ich habe heute das neue Release aufgesetzt. Nice work guys.
Im Wiki habe ich ein FAQ mit folgendem Titel gefunden...
FAQ#can-i-use-a-live-stream-as-background
... hat das mal jemand getestet? Denn so einfach scheint es nicht mehr zu sein. Wenn man besagte Zeile für die URL in der Style.css ändert sollte der Hintergrund auch ohne Motion bereits weiss sein. Aber es wird trotzdem immer der Background bg.jpg geladen. Auch nach dem löschen der Browserdaten. Gibt es vielleicht anderswo noch eine Abhängigkeit über die Theme Einstellung des Adminpanels?

andi34, 2019-10-15:
[image]

Sollte eigentlich reichen

obkram, 2019-10-15:
Funktioniert, danke. Passt bitte jemand den FAQ Bereich entsprechend an? Denn dort steht es noch wie im Issue #20 beschrieben.
Als kleine Info würde ich noch mit aufnehmen das, falls die Steuerung von einem z.B Tablet aus vorgenommen wird, anstellen von 127.0.0.1 die IP Adresse des Verwaltungsrechners eingetragen werden muss. Danke

andi34, 2019-10-15:
Kümmere mich die Tage drum, danke fürs Feedback :)

obkram, 2019-11-09:
Frage zum Wallpaper/ Background ohne einen neuen Issue zu eröffnen. Wäre es grundsätzlich möglich im Bereich +Ordner eine Spalte hinzuzufügen in welcher standartmäßig das Wallpaper bg.jpg Eingetragen ist ( in Abhängigkeit vom ausgewählten Theme ) aber jeder Nutzer seinen Background zum Anlass personalisieren kann? Also ich meine besagte Bereiche in der Style.css mit der Bezeichnung URL.

closed 2019-11-09 by obkram

andi34, 2019-11-09:
@obkram schick mir Mal ne Mail an andi@unlegacy-android.org

magforce, 2020-03-30:
Langsam bin ich echt am Verzweifeln. Ich bekomme das mit dem LiveView einfach nicht hin. Die Vorgehensweise in #20 plus den hier gemachten angaben wollen einfach nicht laufen.

Meine Vorgehensweise:
1. photobooth auf raspbian installiert
2. Konfiguration der OS
3. enable PiCam
4. install motion
5. url(<ip-adresse>)
6. reboot

habe auch versucht die style.css anzupassen, aber auch hier passiert nix im Browser (Chrome).

Was mache ich falsch?

andi34, 2020-03-30:
Live view während des Countdown geht nicht wenn Motion installiert ist. Die Gerätekamera ist somit in Benutzung und kann kein zweites Mal benutzt werden.

Vorschau aus der Gerätekamera ist aber geräteabhängig, heisst: 
Browser am Pi = Pi Camera (falls vorhanden und kein Motion) ,
Browser am Handy = Handykamera (sofern Ausnahme im Browser oder HTTPS und Zugriff auf Kamera erlaubt ist).
Du siehst also an verschiedenen Geräten nicht das selbe Bild wenn die Option für sie Vorschau aus Gerätekamera aktiviert ist.

Wenn Motion installiert ist, dann kannst du den über Motion erzeugten Stream in der aktuellen Version aber über das Admin Panel als Hintergrund im Startbildschirm hinterlegen.

`url(IP-Adresse: Port)`
Siehe auch: https://github.com/andreknieriem/photobooth/wiki/FAQ#can-i-use-a-live-stream-as-background

andi34, 2020-03-30:
Einfacher Test, Motion läuft by default über den Port 8081.
Öffne an deinem Pi Chromium und rufe `http://127.0.0.1:8081` auf. Wenn du kein Bild siehst, dann ist evtl. was an deiner Motion config faul.

magforce, 2020-03-30:
Danke für die Antwort. Hatte nicht wirklich damit gerechnet, dass ich eine bekomme, da das Thema schon geschlossen wurde.

Habe jetzt motion nicht installiert und LiveView beim Countdown läuft mit der PiCam. Aber wie kann ich jetzt mit Hilfe den von motion erzeugten stream als Hintergrund im Starrbildschirm legen?

Ich weiß, dass das Thema bestimmt für dich schon lange Füße hat, aber über Hilfe würde ich mich freuen. 
Was muss ich nach der Installation von Motion wo und wie machen? K.A. wie ich die motion.conf anpassen muss.

andi34, 2020-03-30:
Mit Motion muss ich passen, hatte bisher noch keine Zeit meine PiCam mit Motion zu testen.

Also grundsätzlich wenn Motion läuft dann erreichst du es am Pi selbst im Browser über `http://127.0.0.1:8081`

Ich lass mir Mal ne config zukommen und leite sie dir weiter. Kann nur nen Moment dauern ;)

magforce, 2020-03-30:
> Ich lass mir Mal ne config zukommen und leite sie dir weiter. Kann nur nen Moment dauern ;)

cool, danke

andi34, 2020-03-30:
Kein Thema. In der Zwischenzeit könntest du sonst hier schauen: https://strobelstefan.org/?p=5772
Sieht alles gut erklärt aus :)

magforce, 2020-03-30:
Also ich habe es zum Laufen bekommen. Hatte kurz zuvor schon eine ähnliche Seite auswendig gemacht, die das auch gut beschrieben hat, wie Motion zu installieren und zu konfigurieren ist.

https://www.datenreise.de/raspberry-pi-ueberwachungskamera-livestream/

Jedenfalls bin ich erleichtert, dass es dann doch noch geklappt hat.
Abschließend werde ich jetzt den Pi als AccessPoint einrichten. So kann ich auf einen zusätzlichen Router verzichten. Hat auch bei vorigen Tests schon super funktioniert.

https://github.com/andreknieriem/photobooth/wiki/RPi-as-access-point

Also Danke nochmal und großen Respekt für dieses Projekt.