andreknieriem/photobooth #100: Sytembutton Admin-/ Userpage

obkram, 2019-10-03:
Frage an die Administratoren und Programmierer dieser wundervollen Software. In den meisten Fällen ist die Hardware (Windows PC, RPI, etc. ) in den Boxen selbst verbaut. Da heißt wenn das System mal ein Problem hat heißt es unter Umständen die halbe Box zerlegen ( je nach Bauart ).
Wäre es möglich auf dem jeweiligen lokalen System über die Admin-Page wahlweise ein..
shutown -r, shutdown, sudo reboot oder ein sudo halt ausführen zu lassen?
Sinnvoll wäre zur **Admin-Page** auch eine **User-Page** mit einem abgespeckten Versionsumfang zur eigenen Einstellung durch den Nutzer. Dieser könnte somit auch nach der Nutzung das System kontrolliert herunterfahren. Besser als den Strom zu kappen.

[image]

plin2, 2019-10-03:
Die schnelle Lösung: Leg Dir ein PHP-Script an
**/var/www/html/shutdown.php**
mit
<?php
$cmd="sudo init 0";
$rc = system($cmd, $retval);
?>

obkram, 2019-10-03:
@plin2 Der Tip ist super, vielen Dank.

fossil74, 2019-10-06:
@obkram wie hast du das eingebaut wenn ich fragen darf.

sualko, 2019-10-06:
For security reasons I'm against this feature request. The webserver should not be able to infer with your system is such a way. The better way would be to add a [hw shutdown button to your pi](https://core-electronics.com.au/tutorials/how-to-make-a-safe-shutdown-button-for-raspberry-pi.html) or to use ssh.

andi34, 2019-10-06:
> Die schnelle Lösung: Leg Dir ein PHP-Script an
> **/var/www/html/shutdown.php**
> mit
> `<?php`
> `$cmd="sudo init 0";`
> `$rc = system($cmd, $retval);`
> `?>`
> Das kannst Du dann vom Tablet oder Deinem Smartphone aus aufrufen.

Way to implement it is mentioned if someone needs.

closed 2019-10-06 by andi34