andreknieriem/photobooth #99: Enhancement: Camera profiles and switching between them

plin2, 2019-10-03:
I intend to build a picam into my photobooth and optionally (when I want hight quality) attach a digital camera on top. 

It would be helpful when the photobooth software could detect which cameras are detected (vcgencmd get_camera, gphoto2 --auto-detect), store profiles (the camera dependend options) and allow switching between them. A profile would consist of the following options

- $config['camera_model'] = 'picam';
- $config['use_picamera'] = true;
- $config['crop_width'] = '2592';
- $config['crop_height'] = '1850';
- $config['take_picture']['cmd'] = 'sudo raspistill -n -o $(date +%s) | echo Done';
- $config['take_picture']['msg'] = 'Done';

or

- $config['camera_model'] = 'Canon PowerShot G9;
- $config['use_picamera'] = false;
- $config['crop_width'] = '4000';
- $config['crop_height'] = '2666';
- $config['take_picture']['cmd'] = 'sudo gphoto2 --capture-image-and-download --filename=%s images';
- $config['take_picture']['msg'] = 'New file is in location';

sualko, 2019-10-06:
In my opinion there are more urgent issues than a camera selection dialog. If you need such a thing now, just use two config files and link them depending on which camera you want to use.

plin2, 2019-10-06:
> If you need such a thing now, just use two config files and link them depending on which camera you want to use.

That's what I'm doing right now

> In my opinion there are more urgent issues than a camera selection dialog. 

Consider my enhancement request as a compliment - making a good product even better :-)

andi34, 2019-12-30:
Well, we could make the my.config name adjustable via admin panel. This way you'll be able to switch between different configs.
Maybe also make database file  and mailer list adjustable?